bazmonk/digigurdy-baz #4: Feature: More strings!

bazmonk, 2022-02-26:
Teensy3.5 should support 16 MIDI channels, 8 in, 8 out.

Digigurdy has four string channels, one for the keyclick, and one for the trompette buzz.

Little math here...
* 4 + 1 + 1 = 6
* 8 - 6 = 2
* 2/6 * 100 == 33%

Therefore it looks like it could be _up to a third more awesome_ if we made use of those channels :-)

Adding the extra string and using another channel shouldn't be too hard.  What will be hard is how to handle tuning options for it without it becoming a mess.  Other questions...

* Does the fact that the notes are perfectly in tune detract from the extra presence that multiple strings make?  Would more just get lost in the harmonics or really sound like more strings?  I think MIDI notes can be microtuned... would intentionally making the tuning slightly imperfect yield a more realistic-sounding result?
* Beyond what's possible in a 2-2-2 setup, how are more elaborate gurdies set up?  Research into how they are used may lead to refinements in how to use four drone channels together.

This issue might stick around for a while but it's on my mind.

bazmonk, 2022-02-26:
Something to keep in mind: John Dingley said latency might become an issue.

A more programmatic way to measure latency would be nice...

Gubbledenut, 2022-02-27:
> Teensy3.5 should support 16 MIDI channels, 8 in, 8 out.
>
> Digigurdy has four string channels, one for the keyclick, and one for the trompette buzz.
>
> Little math here...
>
> * 4 + 1 + 1 = 6
> * 8 - 6 = 2
> * [...]

A reason for settling on the 4 string version is not just about latency etc, but also about the quality of the sound itself, and perceived requirement..when less is more when actually playing..having a third chanter would be wonderful sometimes for harms, also paling in C I often also use the G drone too, using 5th drones playing complex bagpipes is common, but not so much on gurdies. By complex bagpipes I mean musette du cour and Northumbrian small pipes, the mussette especially and other complex shuttle drones play in "modes" it gets complicated..however with midi produced sounds its very difficult to recreate the overtones, harmonics etc that come out from reeds and strings..they can also be described as errors and faults that in theory should sound crap, but in practice in particular keys or modes sound incredible....

bazmonk, 2022-03-05:
I have been doing some experimenting with pitch bending the strings very, very slightly out of tune and I think I'm on to something.  Two notes in precise unison blend in too perfectly with each other to have the effect of hearing two strings, but if they're just slightly off (so little that on a physical gurdy you'd still consider that in-tune, a couple cents) it does sound like you get some of that back.

The rub is that it does take cooperation with your synthesizer/app, since it decides what to actually do with the pitch bend information.  I'm setting bs-16i to bend over just 1 semitone up and down, and someone else would have to do the same to get the effect.

Even if I don't do much with extra strings, just for my personal use I am definitely going to create a pitch-bended mode or setting that tweaks the strings off just a little like I'm describing for the existing four strings.