bazmonk/digigurdy-baz #12: Feature: digiharpa support

bazmonk, 2022-02-28:
@Gubbledenut, this is for you.

Now that I know digiharpas also use digigurdy code, I'm keeping that in mind with the rewrite.

Supporting them requires:
* Extra keybox buttons
* Digiharpa-specific tuning options

I'll work on tuning options when it gets to that stage, but in order to support extra buttons, KeyboxButton objects in the code (which represent the pin assingment and how high that button raises a string's note) will be created from an array, that eventually I'll put in a separate configuration file.  From https://github.com/bazmonk/digigurdy-baz/tree/rewrite:

```
int pin_array[] = {-1, 2, 24, 3, 25, 26, 4, 27, 5, 28, 29, 6, 30,
                   7, 31, 8, 32, 33, 18, 34, 19, 35, 20, 36, 37};
```

This is what the digigurdy uses.  -1 is a bogus filler value (it won't get a button object), but for the rest the value is the pin assignment, and the array index is how many half-steps it raises a note.  So like, pin 25 is index 4 so it raises the string by two whole notes.

That way, digiharpa support is just a matter of providing a different array.

Gubbledenut, 2022-02-28:
Excellent, looking at your latest commit, I see I could add extra keys to the pin array.
Will test latest commit today ..it's still saying v  0.3, but obvious it's changed.

bazmonk, 2022-03-01:
This version of the code has no display, no tuning options, just two melody strings, no crank, no buzzing... it compiles and runs and works, but it's very much not done.

I'm also really willy-nilly with the version numbering.  I called the latest v0.5 just to bump it up.  The only version number that's going to really mean something is v1.0: a feature-complete version suitable for end-users.

bazmonk, 2022-03-13:
I'm moving this to a milestone after version 1.0.

Now that i know more of the situation, I'm not sure if proper digiharpa support will exist as an option in the main code.  Since the rewrite, it seems like it will be more like me just being cognizant that digiharpas exist and may use this code.  I've made it a point to not make digiharpa support more difficult, and I think with that it's in a place where these kind of changes are easy to implement.

"Not making porting to digiharpa harder than it needs to be" might be the most "support" it gets in an official capacity.  It's hard for me to make any kind of promise that such a port would work without having one to test with.

Gubbledenut, 2022-03-13:
it's ok, really don't spend any time on it, I can adjust the keymap and tuning for this, when and if we ever get the airbow working..John's onto that now.
Having fun with the extended tenor key box on the digigurdy for now, which means it will work with my harpa layouts..and as I'm the only one making it, and it won't be going into production!!!

bazmonk, 2022-03-15:
I think I'm just going to close this then, as we're good with the keymap adjustment, and we'll address any other specifics later.

closed 2022-03-15 by bazmonk