bazmonk/digigurdy-baz #13: Tunings

Gubbledenut, 2022-03-01:
Suggestions below for menu orders...little more logic, Also suggest a 9. G2/D3 to replace current 9. G3/C4 very often used playing in G on a G/C gurdy
//Suggested G/C Menu Order
         " Select Drone/Tromp: \n" 
         "  2. C2/C3 | 10.G3/G4 \n"
		  " 3. C2/C4 | 5.G2/G3 \n"
         "  1. C2/G2 | 4.G2/C3 \n"
		  " 7. C3/G3 | 6 G2/C4 \n"
		  " 8. C3/G4 | 9.G3/C4 \n"  
		   
		  Sugggest 9 is changed to G2/D3
		  
		//Suggested D/G Menu Order
		 " Select Drone/Tromp: \n"
		 " 2. D2/D3 | 4. G2/D3 \n"
         " 8. D3/D4 | 6. G2/D4 \n"
		 " 1. D2/G2 | 5. G2/G3 \n"
		 " 3. D2/G4 | 9. G3/D4 \n"
		 " 7. D3/G3 | 10. G3/G4 \n"

bazmonk, 2022-03-01:
I can definitely do this.  The rationale with those drone/trompette combos was each one I could think of starting at octave 2 up to octave 4 less than two octaves apart. If G2/[C,D]3 is missing that was my fault.  And they're in a fairly arbitrary order.

It'll be a little while... I'm smack in the middle of rewriting the rest of the code (https://github.com/bazmonk/digigurdy-baz/tree/rewrite), and I want to finish it to have a cleaner environment to work with.  That version doesn't even have the tuning menus yet.

If I don't make this change when I re-implement the tuning menus, it'll be the first change to v1.0.

At the rate I'm going, I think this will be in a couple weeks or so.

Gubbledenut, 2022-03-01:
I'll take a look... This below is an updated pinout for all the Digi things I've based on the standard 24 key digigurdy keyboard. Note the DigiTenor goes  7 keys below the digigurdy while  the DigiMora and DigiHarpa go above.
For reference I've also included the SampleBox Wav numbers... 

//XXXXXSTANDARD 24 KEY Digigurdy SetupXXXXX
//        Teensy Pin|| MIDI Value G/C|| Note G/C  ||Note D/G  || MIDI Value D/G
//MIDI Values for Keyed notes are also the SampleBox Wav No. (Wav file available for keyed notes  50 to 105 D3 to D  )
int aa;//     37    91         G6	D6   86  
int bb;//     36    89         F6	C6   84
int cc;//     20    90         F6#	C6#  85
int dd;//     35    88         E6	B5   83
int ee;//     19    87         D6#	A5#  82
int ff;//     34    86         D6	A5   81
int gg;//     18    85         C6# 	G#5	 80
int hh;//     33    84         C6	G5   79
int ii;//     32    83         B5	F5#  78
int jj;//     8     82         A5#	F5   77
int kk;//     31    81         A5	E5   76
int ll;//     7     80         G5#	D5#  75
int mm;//     30    79         G	D5   74
int nn;//     6     78         F5#	C5#  73
int oo;//     29    77         F5	C5   72
int pp;//     28    76         E5	B4   71
int qq;//     5     75         D5#	A4#  70
int rr;//     27    74         D5	A4   69
int ss;//     4     73         C5#	G4#  68
int tt;//     26    72         C5	G4   67
int uu;//     25    71         B4	F4#  66
int vv;//     3     70         A4#	F4   65
int ww;//     24    69         A4	E4   64
int xx;//     2     68         G4#	D4#  63
//    --     		67         G4	D4   62       Open string

//XXXX Tenor Digigurdy Extra Notes XXXX

//DJ Added TenorDigiGurdy notes yy to qq1 for extra low Tenor Keys 
//To extend down from Standard 24 key Digigurdy to 31 Keys Tenor 
		Teensy Pin|| MIDI Value G/C|| Note G/C  ||Note D/G  || MIDI Value D/G
int yy;//     48    67         G4	D4  62	Also Open Chanter note on standard 24 keyboard
int zz;//     49    66         F#4	C#4	61
int jj1;//    50    65         F4	C4	60
int kk1;//    51    64         E4	B3	59
int ll1;//    52    63         Eb4	Bb3	58
int mm1;//    53    62         D4	A3	57
int nn1;//    54    61         C#4	G#3	56
//Open string 		60         C4 	G3	55

//Starting Tunings for Tenor G Cc G for a G/C and D Gg D for a D/G

//XXXXDigiHarpa Extra Notes XXXX

//DJ Added DigiHarpa notes yy to rr1 for extra high notes 
//To extend up from 24 key Digigurdy 35 keys DigiHarpa
//See Keymap for actual layout, but this doesn't effect code.
Keys Below added for DigiHarpa
int yy;//     38            92         G#6          
int zz;//     14            93         A6  
int jj1;//    43            94         A#6 
int kk1;//    44            95         B6
int ll1;//    45            96         C7
int mm1;//    46            97         C#7
int nn1;//    47            98         D7
int oo1;//    48            99         Eb7
int pp1;//    49            100        E7
int qq1;//    50            101        F7
int rr1;//    51            102        F#7

Harpa Open Strings Standard Tuning
chan1midi = 81;  //  A5
chan2midi =	72;  //  C5 (or D5)
chan3midi = 67;  //  G4
chan4midi = 60;  //  C4

//XXXX DigiMora Extra Notes XXXX

// DJ Added DigiMora notes yy to zz for extra high note
// To extend up from 24 key Digigurdy to 26 keys DigiMora
int yy;//     38            92         G#6            
int zz;//     14            93         A6
/Moraharpa tuning is a bit in the aire right now but 3 strings this can wait!

Mods to Wav files for SampleBox Melody Strings stay the same
Trompette file correct MIDI number          Trompette file new substituted wav file number (MIDI Value -35)
MIDI Value || SampleBox Wav No.||Note

//   60  25	C4
//   62  27	D4
//   64  29	E4
//   67  32	G4

Drone file correct MIDI number   New substituted renumbered value (MIDI Value -35)
MIDI Value || SampleBox Wav No.||Note
//   36  1	C2
//   38  3	D2
//   43  8	G2
//   45  10	A
//   48  13	C3
//   50  15	D3
//   57  22	A3

renumber the buzz sound files as follows
//   Equivalent Trompette MIDI value  New substituted renumbered value (MIDI Value -27)
MIDI Value || SampleBox Wav No.||Note
//   60  33	C4
//   62  35	D4
//   64  37	E4
//   67  40 G4

bazmonk, 2022-03-06:
Closing this: it tuning works differently in the re-write.  Will keep this file for digiharpa support plans, though.

closed 2022-03-06 by bazmonk