bazmonk/digigurdy-baz #20: Feature: DoReMi, alternate notations

bazmonk, 2022-03-06:
@gubbledenut this is for you!

Set up the config flags (compile-time) and some alternate NoteNum and LongNoteNum arrays for displaying DoReMi-style notation on-screen.

Another suggestion was to possibly remove the octave number in an alternate display (if I read that correctly?).  At any rate, with the re-write, I'm not sure...  It might be really hard to work with the menus without displaying the octave somehow.

bazmonk, 2022-03-07:
I’m not sure how octaves are represented in solfège notation.  E.g. C4 = Do… Is C3 just also “Do”, or is ‘Do3’ or `Do’’’` used?  To replace the scientific notation (C4, D#5, etc.) on the display, the values should really identify the octave somehow.  I must be searching for the wrong thing because I’m having trouble finding info on line on how this is done in fixed-Do solfège.

bazmonk, 2022-03-07:
Another concern is that the “long” version of the note strings are 7 characters max, so the replacement has to fit in that as well.

bazmonk, 2022-03-07:
In solfège are the flats usually given as well?  Do#/Re♭?  My concern with including both is that “C♯/D♭, D♯/Re♭“ is too long.  

So you’re saying leave ABC for the tuning but just display the note being played in solfège?  I can do that…

> the issue of the octave shown in the graphic

Ah yes, the bitmaps.  Have not considered how/if to add them back in.

> I know someone asked for it…

And now someone’s just asking for it to go away :-). I’ll look into an option for hiding it during play, or a compact view during play.  When not-playing I don’t see a reason to have to hit the capo or something to see it.  It’s not like the screen has something better to display when nothing is happening, you know?

Gubbledenut, 2022-03-07:
Already answered this, same in DoReMi as ABC with sharps and flats most commonly used rather in gc or dg tunings and keys.
C#,Eb,F#,Bb should be sufficient

bazmonk, 2022-03-07:
Ok.  I’ve got some ideas.  I’ll update this when there’s something to show you.

Gubbledenut, 2022-03-11:
Thinking aloud now, after playing around a bit more with your code.
I think the DoReMi formatting can be done with another array ....and amalgamated if required rather than "force" it on everyone..... let me play a little more with the code

bazmonk, 2022-04-05:
I'm sorry because I know you've answered this before, but I'm still a little unclear how you want this to work.  Perhaps you could help me?

Below are the two arrays that guide the (text, bitmaps I handle after this) display of notes.  NoteNum should each be 4 characters max (I'll have to tweak a few screens but that's ok), and LongNoteNum _needs_ to be 7 characters max each.

Could you just reply here with an example of what you'd like for both of these?  Like, just one octave of the short and long variety and I'll fill in the rest?  Since you want to favor some sharps and flats over others--and since this is your request--I want to make the DoReMi mode exactly how you want it.

With that I'll create the option to switch at startup (in the other options menu), and see what I can do for bitmaps.  I really hope I can fit it in the same general setup I have the current bitmaps with minimal changes...

Thanks!

```
std::string NoteNum[] = {
  "C-1", "C#-1", "D-1", "D#-1", "E-1", "F-1", "#F-1", "G-1", "G#-1", "A-1", "#A-1", "B-1",
  "C0", "C#0", "D0", "D#0", "E0", "F0", "F#0", "G0", "G#0", "A0", "A#0", "B0",
  "C1", "C#1", "D1", "D#1", "E1", "F1", "F#1", "G1", "G#1", "A1", "A#1", "B1",
  "C2", "C#2", "D2", "D#2", "E2", "F2", "F#2", "G2", "G#2", "A2", "A#2", "B2",
  "C3", "C#3", "D3", "D#3", "E3", "F3", "F#3", "G3", "G#3", "A3", "A#3", "B3",
  "C4", "C#4", "D4", "D#4", "E4", "F4", "F#4", "G4", "G#4", "A4", "A#4", "B4",
  "C5", "C#5", "D5", "D#5", "E5", "F5", "F#5", "G5", "G#5", "A5", "A#5", "B5",
  "C6", "C#6", "D6", "D#6", "E6", "F6", "F#6", "G6", "G#6", "A6", "A#6", "B6",
  "C7", "C#7", "D7", "D#7", "E7", "F7", "F#7", "G7", "G#7", "A7", "A#7", "B7",
  "C8", "C#8", "D8", "D#8", "E8", "F8", "F#8", "G8", "G#8", "A8", "A#8", "B8",
  "C9", "C#9", "D9", "D#9", "E9", "F9", "F#9", "G9"
};

// This is a version of the above but with flats listed as well.
std::string LongNoteNum[] = {
  " EMPTY ", "C#-1/Db-1", "  D-1  ", "D#-1/Eb-1", "  E-1  ", "  F-1  ", "F#-1/Gb-1", "  G-1  ", "G#-1/Ab-1", "  A-1  ", "A#-1/Bb-1", "  B-1  ",
  "  C0   ", "C#0/Db0", "  D0   ", "D#0/Eb0", "  E0   ", "  F0   ", "F#0/Gb0", "  G0   ", "G#0/Ab0", "  A0   ", "A#0/Bb0", "  B0   ",
  "  C1   ", "C#1/Db1", "  D1   ", "D#1/Eb1", "  E1   ", "  F1   ", "F#1/Gb1", "  G1   ", "G#1/Ab1", "  A1   ", "A#1/Bb1", "  B1   ",
  "  C2   ", "C#2/Db2", "  D2   ", "D#2/Eb2", "  E2   ", "  F2   ", "F#2/Gb2", "  G2   ", "G#2/Ab2", "  A2   ", "A#2/Bb2", "  B2   ",
  "  C3   ", "C#3/Db3", "  D3   ", "D#3/Eb3", "  E3   ", "  F3   ", "F#3/Gb3", "  G3   ", "G#3/Ab3", "  A3   ", "A#3/Bb3", "  B3   ",
  "  C4   ", "C#4/Db4", "  D4   ", "D#4/Eb4", "  E4   ", "  F4   ", "F#4/Gb4", "  G4   ", "G#4/Ab4", "  A4   ", "A#4/Bb4", "  B4   ",
  "  C5   ", "C#5/Db5", "  D5   ", "D#5/Eb5", "  E5   ", "  F5   ", "F#5/Gb5", "  G5   ", "G#5/Ab5", "  A5   ", "A#5/Bb5", "  B5   ",
  "  C6   ", "C#6/Db6", "  D6   ", "D#6/Eb6", "  E6   ", "  F6   ", "F#6/Gb6", "  G6   ", "G#6/Ab6", "  A6   ", "A#6/Bb6", "  B6   ",
  "  C7   ", "C#7/Db7", "  D7   ", "D#7/Eb7", "  E7   ", "  F7   ", "F#7/Gb7", "  G7   ", "G#7/Ab7", "  A7   ", "A#7/Bb7", "  B7   ",
  "  C8   ", "C#8/Db8", "  D8   ", "D#8/Eb8", "  E8   ", "  F8   ", "F#8/Gb8", "  G8   ", "G#8/Ab8", "  A8   ", "A#8/Bb8", "  B8   ",
  "  C9   ", "C#9/Db9", "  D9   ", "D#9/Eb9", "  E9   ", "  F9   ", "F#9/Gb9", "  G9   "
};
```

Gubbledenut, 2022-04-05:
Think this is right....
ideally having both the universal ABC and DoReMi is my preference...or the graphic...it's good to learn both especially playing European music comes in handy.
[DoReMi.txt](https://github.com/bazmonk/digigurdy-baz/files/8415723/DoReMi.txt)
it doesn't past too well
std::string NoteNum[] = {
  "Do-1", "Do#-1", "Re-1", "Re#-1", "Mi-1", "Fa-1", "#Fa-1", "Sol-1", "Sol#-1", "La-1", "#La-1", "Si-1",
  "Do0", "Do#0", "Re0", "Re#0", "Mi0", "Fa0", "Fa#0", "Sol0", "Sol#0", "La0", "La#0", "Si0",
  "Do1", "Do#1", "Re1", "Re#1", "Mi1", "Fa1", "Fa#1", "Sol1", "Sol#1", "La1", "La#1", "Si1",
  "Do2", "Do#2", "Re2", "Re#2", "Mi2", "Fa2", "Fa#2", "Sol2", "Sol#2", "La2", "La#2", "Si2",
  "Do3", "Do#3", "Re3", "Re#3", "Mi3", "Fa3", "Fa#3", "Sol3", "Sol#3", "La3", "La#3", "Si3",
  "Do4", "Do#4", "Re4", "Re#4", "Mi4", "Fa4", "Fa#4", "Sol4", "Sol#4", "La4", "La#4", "Si4",
  "Do5", "Do#5", "Re5", "Re#5", "Mi5", "Fa5", "Fa#5", "Sol5", "Sol#5", "La5", "La#5", "Si5",
  "Do6", "Do#6", "Re6", "Re#6", "Mi6", "Fa6", "Fa#6", "Sol6", "Sol#6", "La6", "La#6", "Si6",
  "Do7", "Do#7", "Re7", "Re#7", "Mi7", "Fa7", "Fa#7", "Sol7", "Sol#7", "La7", "La#7", "Si7",
  "Do8", "Do#8", "Re8", "Re#8", "Mi8", "Fa8", "Fa#8", "Sol8", "Sol#8", "La8", "La#8", "Si8",
  "Do9", "Do#9", "Re9", "Re#9", "Mi9", "Fa9", "Fa#9", "Sol9"
};

// This is a version of the above but with flats listed as well.
std::string LongNoteNum[] = {
  " MiMPTY ", "Do#-1/Reb-1", "  Re-1  ", "Re#-1/Mib-1", "  Mi-1  ", "  Fa-1  ", "Fa#-1/Solb-1", "  Sol-1  ", "Sol#-1/Lab-1", "  La-1  ", "La#-1/Sib-1", "  Si-1  ",
  "  Do0   ", "Do#0/Reb0", "  Re0   ", "Re#0/Mib0", "  Mi0   ", "  Fa0   ", "Fa#0/Solb0", "  Sol0   ", "Sol#0/Lab0", "  La0   ", "La#0/Sib0", "  Si0   ",
  "  Do1   ", "Do#1/Reb1", "  Re1   ", "Re#1/Mib1", "  Mi1   ", "  Fa1   ", "Fa#1/Solb1", "  Sol1   ", "Sol#1/Lab1", "  La1   ", "La#1/Sib1", "  Si1   ",
  "  Do2   ", "Do#2/Reb2", "  Re2   ", "Re#2/Mib2", "  Mi2   ", "  Fa2   ", "Fa#2/Solb2", "  Sol2   ", "Sol#2/Lab2", "  La2   ", "La#2/Sib2", "  Si2   ",
  "  Do3   ", "Do#3/Reb3", "  Re3   ", "Re#3/Mib3", "  Mi3   ", "  Fa3   ", "Fa#3/Solb3", "  Sol3   ", "Sol#3/Lab3", "  La3   ", "La#3/Sib3", "  Si3   ",
  "  Do4   ", "Do#4/Reb4", "  Re4   ", "Re#4/Mib4", "  Mi4   ", "  Fa4   ", "Fa#4/Solb4", "  Sol4   ", "Sol#4/Lab4", "  La4   ", "La#4/Sib4", "  Si4   ",
  "  Do5   ", "Do#5/Reb5", "  Re5   ", "Re#5/Mib5", "  Mi5   ", "  Fa5   ", "Fa#5/Solb5", "  Sol5   ", "Sol#5/Lab5", "  La5   ", "La#5/Sib5", "  Si5   ",
  "  Do6   ", "Do#6/Reb6", "  Re6   ", "Re#6/Mib6", "  Mi6   ", "  Fa6   ", "Fa#6/Solb6", "  Sol6   ", "Sol#6/Lab6", "  La6   ", "La#6/Sib6", "  Si6   ",
  "  Do7   ", "Do#7/Reb7", "  Re7   ", "Re#7/Mib7", "  Mi7   ", "  Fa7   ", "Fa#7/Solb7", "  Sol7   ", "Sol#7/Lab7", "  La7   ", "La#7/Sib7", "  Si7   ",
  "  Do8   ", "Do#8/Reb8", "  Re8   ", "Re#8/Mib8", "  Mi8   ", "  Fa8   ", "Fa#8/Solb8", "  Sol8   ", "Sol#8/Lab8", "  La8   ", "La#8/Sib8", "  Si8   ",
  "  Do9   ", "Do#9/Reb9", "  Re9   ", "Re#9/Mib9", "  Mi9   ", "  Fa9   ", "Fa#9/Solb9", "  Sol9   "
};

bazmonk, 2022-04-05:
That’s the prob… LongNoteNum needs to be seven characters max.  These are 9.

You were saying that most of the time certain sharps/flats are used more often than their counterpart.  Could we make LongNoteNum short enough by only including one or the other?

Gubbledenut, 2022-04-05:
I can do that, will also make up a list with universal and DoReMi notes in it...

bazmonk, 2022-04-06:
I'll probably leave the ABC ones as is because 1) I've already made space to accommodate it, and 2) I like it personally.

But maybe an "abbreviated" list in ABC could be a third option...

Gubbledenut, 2022-04-07:
this, but not the sharps flats ones just yet...

std::string NoteNum[] = {
  "C/Do-1", "C/Do#-1", "D/Re-1", "D/Re#-1", "E/Mi-1", "F/Fa-1", "#F/Fa-1", "G/Sol-1", "G/Sol#-1", "A/La-1", "#A/La-1", "B/Si-1",
  "C/Do0", "C/Do#0", "D/Re0", "D/Re#0", "E/Mi0", "F/Fa0", "F/Fa#0", "G/Sol0", "G/Sol#0", "A/La0", "A/La#0", "B/Si0",
  "C/Do1", "C/Do#1", "D/Re1", "D/Re#1", "E/Mi1", "F/Fa1", "F/Fa#1", "G/Sol1", "G/Sol#1", "A/La1", "A/La#1", "B/Si1",
  "C/Do2", "C/Do#2", "D/Re2", "D/Re#2", "E/Mi2", "F/Fa2", "F/Fa#2", "G/Sol2", "G/Sol#2", "A/La2", "A/La#2", "B/Si2",
  "C/Do3", "C/Do#3", "D/Re3", "D/Re#3", "E/Mi3", "F/Fa3", "F/Fa#3", "G/Sol3", "G/Sol#3", "A/La3", "A/La#3", "B/Si3",
  "C/Do4", "C/Do#4", "D/Re4", "D/Re#4", "E/Mi4", "F/Fa4", "F/Fa#4", "G/Sol4", "G/Sol#4", "A/La4", "A/La#4", "B/Si4",
  "C/Do5", "C/Do#5", "D/Re5", "D/Re#5", "E/Mi5", "F/Fa5", "F/Fa#5", "G/Sol5", "G/Sol#5", "A/La5", "A/La#5", "B/Si5",
  "C/Do6", "C/Do#6", "D/Re6", "D/Re#6", "E/Mi6", "F/Fa6", "F/Fa#6", "G/Sol6", "G/Sol#6", "A/La6", "A/La#6", "B/Si6",
  "C/Do7", "C/Do#7", "D/Re7", "D/Re#7", "E/Mi7", "F/Fa7", "F/Fa#7", "G/Sol7", "G/Sol#7", "A/La7", "A/La#7", "B/Si7",
  "C/Do8", "C/Do#8", "D/Re8", "D/Re#8", "E/Mi8", "F/Fa8", "F/Fa#8", "G/Sol8", "G/Sol#8", "A/La8", "A/La#8", "B/Si8",
  "C/Do9", "C/Do#9", "D/Re9", "D/Re#9", "E/Mi9", "F/Fa9", "F/Fa#9", "G/Sol9"
};

Gubbledenut, 2022-04-07:
Yes , i think if it’s meant as a learning aid then good idea to include the option for dual “language “ .

bazmonk, 2022-11-03:
Just kicking this can down the street for now.  When the new display is displaying, it'll be a good time to revisit this with fresh (and more C++-skilled) eyes now that it's been months.