bazmonk/digigurdy-baz #21: not playing or displaying correctly

Gubbledenut, 2022-03-06:
ver 0.9 No matter which tuning I select, when I go to play it shows the open string as C#-1 and almost random tuning display 
EG G/C **selection each time C Gg C tuning (Tromp, Low chanter, High Chanter, Drone) Shows 
Tpose +67 Capo +7
hi melody F#0
low Melody #F-1
Trompette B5
Drone B2
I've tried other tunings with similar results
Works perfectly on parsed ver117

bazmonk, 2022-03-06:
Is this on a regular “stock” digigurdy?  Are you modifying the code at all?

Gubbledenut, 2022-03-06:
My digigurdy are 100% identical in design  to the “original digigurdy” I have not modified the code. 
The only difference is I have a 4.1 teensy in this one. Which is the same pin for pin on the analogue and digital pins. I can also try it on the “test rig” which uses a ver 1 pcb and jumper leads, with a 3.5 and 3.6 . I may also clear the eeprom from my working digigurdy to see if that helps

bazmonk, 2022-03-07:
EEPROM is a good idea.  I'm just blindly reading/writing to it and I'm not
sure what kind of shenanigans it can get into.

If the 4.1 is the same in pin assignment, I'm sort of at a loss of how it
ends up with bizarre tunings like C#-1.  Even if you strategically choose
the right tuning options, there's no way to get it to make a C#-1 note, you
know?  Could the source code itself have gotten formatted strangely?  Is
this a Unix/Win carriage return kind of problem perhaps?

bazmonk, 2022-03-07:
@Gubbledenut Does the Teensy 4.1 fit right in a “original” digigurdy where the 3.5 is?  If it’s the same form-factor, I’ll buy a 4.1, just to have the faster unit and to have one to test against.

Gubbledenut, 2022-03-07:
SUCCESS.... I cleared the eprom and the software is working...I should have thought earlier, we've had similar issues a few times, when there have been major changes to the code..even though it shouldn't look to the eprom unless its told to there seems to be some spill???
Maybe just an idea to suggest using this file when installing It over the previous 117 or earlier code...but be aware it will effectively brick the whole thing until new code is uploaded!!!!!! 
[eeprom_clear.ino.zip](https://github.com/bazmonk/digigurdy-baz/files/8195623/eeprom_clear.ino.zip)

as for the 4.1 yes it does..the pins on the pinout are the same, but the ones on the board itself are in a different place for the higher numbers. Also the 4 series boards don't have cap touch...I have also been using mostly 3.6 boards which are about the same price as the 4.1 but with cap touch.and even better on the shorter boards such as the LC/3.2/4.0 the pinouts are also the same unto pin 12 on the one side..so you can use them in place of the longer boards if the need arises and you don't need all the pins of the longer 3.5/3.6/4.1 ...but we do so ignore that

bazmonk, 2022-03-07:
Awesome!  That’s great to hear.  I can’t say I still understand how the EEPROM caused this problem exactly.  The saved settings aren’t part of the tuning menu (it starts with set values not stored in EEPROM).  I get that if it had other data stored it’d create bizarre tunings, but how that was persistent even after trying new tunings I’m unsure.

I’ll check out this clear code and include it in the repository for others.  Thanks for figuring it out and reporting back!

I have DoReMi questions for you in the other issue :-)

closed 2022-03-09 by bazmonk