bazmonk/digigurdy-baz #22: Looking forward… v2.0

bazmonk, 2022-03-07:
V1.0 is coming up soon, time to start thinking about where to take it from here…

- [ ] Code improvement: In re-writing the code I created classes like a good boy for many of the gurdy functions, but a lot of the logic is still bare in the loop() and setup().  Some things appear in odd places in the code for the sake of being defined in the right order.  This can all be a little better.  I’d like to extend what the HurdyGurdy() class does to manage more of the playing logic.  Before it gets out of hand I wanna tighten it up some more.
- [ ] More strings: We’ve got two more channels.  I want them to be optional drones OR chanters.  I’ve done more playing around with scale tuning on bs-16i and I think they can be useful.
- [ ] Improved menu interface: Besides tuning, the menu could be more sophisticated.  Multiple save slots.  String on/off functions.  MIDI volume adjustment.  It could handle a lot more with some planning.
- [ ] “Official” Teensy 4.1 support.  4.1 should be able to work as a drop-in replacement on DigiGurdies for the 3.5 they’re normally built with.  I’ve got one report that this code works on it, and I ordered my own unit to test.  This will not only give makers more flexibility, but users can more easily replace their units, and in the future the improved speed and power of the 4.1 may come in handy.

I’ll flesh this out more later, just getting thoughts down on “paper”.

Gubbledenut, 2022-03-07:
Do pleaser allow an option to just select the original 5 tunings menu as well as the complex one, it's stillnpretty adaptable with the capo and octave keys

bazmonk, 2022-03-07:
@Gubbledenut I probably won’t implement the original menu verbatim, but I can incorporate a “presets” screen that’s easy to access and provides the original tunings.

I was thinking of some number of presets, that would come pre-populated with the original tuning choices.  Users would have an option to overwrite them if they have a different one they want quickly available, and there would be an option to reset the slots to those original presets.  That should give the ease of the original interface (preset options will be quickly accessible with about the same number of key presses to get there), but the flexibility of the new one.

I’m also probably not going to include an option for the octave keys to act as octave keys again, since the transposing feature can accomplish the same thing (max range is +/- one octave) with far more flexibility, and the higher-octave tuning itself is also probably available in the tuning menu.