bazmonk/digigurdy-baz #24: New possibility

lune36400, 2022-03-08:
Hello Basil,
I made a carbon hg, and I added to the standard keyboard, John's dg mechanism with switches, but it takes too much thickness, and above all I would like to be able to adapt it to the other hgs of other players , that's why I started but not yet finished the prototypes of the pcbs, with 25 and 31 keys.
For future use, knowing that I have no motor on the crank, so no detection, no knob needed.
Need:
-Possibility to play with the current DG like the version of John's software or your new software by pressing the gaming button, and thus having the open string active, and the choice of "with or without" drone (included in the software or on Bismark 16.i)
-Possibility by pressing a new button to be able to play without the drones or not (modifiable), and with an additional key on the key box which replaces the lowest note (open string), and thus to have the DG in accompaniment of the game of the HG, the open string will come from the HG, (or from the additional key) and the other notes played would come from the DG and the HG, and therefore to complete the music because with the DG have little add other instruments by other soundfonts.

I had bypassed John's software by lowering the open string to zero, and added a G4 key, on pin 38, it worked, we always want to do better, currently I have to change the software according to the use.

I hope to make myself understood, I am available for more information.
Greetings
Herve

bazmonk, 2022-03-08:
Hi Herve, thanks for opening the issue for this.

* Possiblity to play with/without drones: with a full-featured synthesizer like bs-16i, what I have been doing is simply cutting the volume to the appropriate channel when I wanted to simulate having a drone muted.  But I realize that’s not always an option, so I do have plans to add a menu screen you can call up during play to turn the drone/trompette (the non-melody strings) off.  So yes, that will be an option again soon.
* A new button specifically for tuning on/off drones.  I don’t have have that button obviously, but if you wanted to add such a button I can show you what you’d want to modify in my code to use it.
* A “zero-note” key in the keybox: this should be possible by filling in the pin_array with a valid pin in the first spot, and then modifying the loop that creates the keybox to use it.  Right now when I create the keybox key objects, I ignore the first element of the pin_array since that’s the open string, but everything should work if there was a zero key.

Am I understanding the requests correctly?  If so, I will write up some mock code to show you how you could achieve these things.

-Baz

bazmonk, 2022-03-08:
Also, for a version that never has a crank: I see three options…

* Leave it in place.  It should simply not detect the crank and then work fine.
* Overwrite the crank-detection method to simply report false and not even try to detect.
* Rip out the crank logic entirely.  This is the cleanest option but requires the most modification from the vanilla codebase.

lune36400, 2022-03-08:
Thanks,
For me for the moment it's how many buttons i must add on the panel for the moment i use G4# F6# D6#? and button pin 21 22 23 like code of John, and i cut the ground to stop all the electronic hall effect to close when you want to stop quickly the play

lune36400, 2022-03-08:
I can leave a copy off the pcb somewhere, that you can see what i do. i use a free software easyEDA to design the pcb

bazmonk, 2022-03-08:
> For me for the moment it's how many buttons i must add on the panel for
> the moment i use G4# F6# D6#? and button pin 21 22 23 like code of John,
>
Sorry, I'm not quite sure what that means.

> and i cut the ground to stop all the electronic hall effect to close when
> you want to stop quickly the play
>
I think that would be interpreted by the bounce library as a button release
so I *think* that should still work.

Do you have a version of John Dingley's code that you've modified to work
with your project?  Seeing how that differs would be helpful.

> Message ID: ***@***.***>
>

lune36400, 2022-03-08:
i don't know if it's possible to put picture on github? i can put it on

lune36400, 2022-03-08:
//Sorry, I'm not quite sure what that means.

I do a pcb for modify the actual pcb, more thon and hall effect.
 i cut the ground, because for me, it's a solution to stop all the song when i want only the hurdy gurdy sound, and not dg sound, and because i don't understand arduino or a little.
i use the last one version of john but i only add a new key for g4 and the volume of openstring =0  

how i can live the code on github for you can see it?

completly neebee, but fun for doing all of that.

bazmonk, 2022-03-08:
This link explains how to fork a repository.  You're basically just cloning a copy of it for yourself.

https://docs.github.com/en/get-started/quickstart/fork-a-repo

There's lots of ways to share files, but once you fork this repo you can also use it to hold some images to share.

bazmonk, 2022-03-09:
> Pins 21 22 and 23 are the capo , and octave up or down buttons, I also
> duplicate pins to buttons for the X and O keys giving my button patches a
> total of 5 buttons.
> I might give it a go modding the code to allow for the extra 7 tenor keys,
> my only concern then is adjusting the tuning correctly…might be possible
> with your existing menu options, the 0.9 seems stable enough, but will
> adjust latest 0.92 .

7 extra keybox keys, are they chromatic like a hurdy gurdy keybox?  Is it
just a longer one?  If so, adjusting the key array and size is literally
all it should take to support those keys:

```// Eventually I'll move this to a header, but the pin_array[] index here
represents
// the MIDI note offset, and the value is the corresponding teensy pin.
// This defines which physical keys are part of the "keybox" and what order
they're in.
//
// NOTE: There's no key that produces 0 offset (an "open" string),
// so the first element is bogus.  It gets skipped entirely.
int pin_array[] = {-1, 2, 24, 3, 25, 26, 4, 27, 5, 28, 29, 6, 30,
                   7, 31, 8, 32, 33, 18, 34, 19, 35, 36, 20, 37};

// This is literally just the size of the above array minus one.  I need
this as a const to
// declare the KeyboxButton array later on... or I just don't know enough
C++ to know how to
// *not* need it ;-)
const int num_keys = 24;
```

If the extra X O buttons you have go to their own pins, you'd have to
duplicate code to use them in addition to what I have.  You could change
the pins in the code to *replace* what I have for X and O keys.

Moving pins (say you use different pins for your capo button) is just
changing that in the code where it's initialized.

Feel free to mod it all you want.  I'm working on 0.9.5 right now, it's
changing a lot of the menu beyond the tuning portion.  Might want to wait
just a couple more days for it to be more stable.

Gubbledenut, 2022-03-09:
the extra keys are an add-on, connected by plugs, so can take off and run a 24 key box too
good pint about the X and O keys on my button box, everything else on pins,21.22.23 will be fine and the selection keys 1- 8 should just shift along so it recognises the extension ones, once I've adjusted the code..may still play with 0.92 copy, looking forward to 0.95
[image]

bazmonk, 2022-03-09:
Ah, ok.  Rearranging the pin_array and increasing the num_keys should work
fine.  They're just the 7 lowest keys now and everything else is shifted up
7 spots in the array.

Or do you want the tenor keys to go *lower* than the open string?  If *that* is
the case, heaver modification is needed, but I didn't do anything to get in
your way.  You just need to code in starting up keys with a negative
offset.  The rest of the code should play along with that no-problem.

Gubbledenut, 2022-03-09:
this works....still needed to clear the eeprom, I got the stall at crank on, each time I change anything???
But it is working, may post a video later ..the tuning options work for it too, as we thought it's just the hard wired x and o that are now in the wrong place. When I get around to the harpa it will be more of an issue, the keys are in different physical positions for the same relative notes and teensy pins, but for now this is great.

// NOTE: There's no key that produces 0 offset (an "open" string),
// so the first element is bogus.  It gets skipped entirely.
/*int pin_array[] = {-1, 2, 24, 3, 25, 26, 4, 27, 5, 28, 29, 6, 30,
                   7, 31, 8, 32, 33, 18, 34, 19, 35, 36, 20, 37};*/
//DJ Array to include Tenor pins for 4.1
 int pin_array[] = {-1, 54, 53, 52, 51, 50, 49, 48, 2, 24, 3, 25, 26, 4, 27, 5, 28, 29, 6, 30,
                   7, 31, 8, 32, 33, 18, 34, 19, 35, 36, 20, 37};

// This is literally just the size of the above array minus one.  I need this as a const to
// declare the KeyboxButton array later on... or I just don't know enough C++ to know how to
// *not* need it ;-)
//const int num_keys = 24;
//DJ Tenor keybaord for 4.1
const int num_keys = 31;

bazmonk, 2022-03-09:
v0.9.5 is there now in the main branch.  It's got an option on start-up to
clear the EEPROM on first-time install (under "Other Options").

Stuck on crank-on: that was a bug cleared up somewhere in that branch.  I
created that problem and fixed it in just the last few hours.  The version
on the main branch shouldn't have that problem... let me know if it does.
This one should work pretty clean after a final EEPROM wipe (from this
point on, I won't modify how I use EEPROM, and that should no longer be
necessary).

Gubbledenut, 2022-03-09:
You forgot to change the ver No at the top of the code. The menu graphic is right though..I’ve changed one of the tunings to the tenor C Cc C tuning and added the extra pins..seems to install and work ok ….

bazmonk, 2022-03-09:
Thx... I'm referring to the version in too many places.  I'll make a
#define to make this easier.

lune36400, 2022-03-10:
bonsoir, voici une copie de mon pcb, je pense que vous pouvez le telecharger avec le logiciel esayEda, assez facile a utiliser.

pcb key 350 et 430mm et pcb   support teensy 3.5 + ecran + clavier bouton, les 3 elements peuvent se détacher pour les placer ou ont veux, les sorties Din pour le bluetooth, ont été créer pour etre montés par la suite.

Merci pour vos commentaires.

[Gerber_reglette a fabriquer.zip](https://github.com/bazmonk/digigurdy-baz/files/8225820/Gerber_reglette.a.fabriquer.zip)

lune36400, 2022-03-12:
Hi,
question, possible or not?

* on the same push button, with 2 way, to put 2 key on it , but separatly like this
[image]
 for exemple  ;  * X + O = menu pause, enregistrer/charger le réglage, activer/drones   on the same button ?

* on the soft a possibility to play, like piano possible to play 2-3-4  notes in same times to do "accord" on a optio by button or in menu......

bazmonk, 2022-03-12:
I think such a button would work, but I know much less about the hardware aspect of this.  I don’t see why it wouldn’t.

If I ever supported polyphonic playing on the keybox it would be a long way away.  Another person mentioned this as a way to emulate Nigel Eaton’s two-keybox gurdy… it can’t do that.  You’ll get a multi-note effect and ability to make triad chords, but it won’t sound like Nigel’s does because his second keybox hits a different string.  You would be getting the same sound on top, not a differentiable melody and separate harmony underneath.

A second hesitation is the technical aspect of how it would work.  If it supports, for example, three simultaneous keys, when you press a 4th, which of the already-playing keys turns off?  If we just say “the three *latest* keypresses get sound, and the oldest key press deactivates when a new one plays”, you’ll run into problems in actual play.  The gurdy can’t know which of the keypresses is supposed to be the harmony you’re playing on top of or vide versa.  

My other hesitation is that it’s starting to be a wholesale departure from emulating a hurdy gurdy.  It’s just a MIDI keyboard now, and other people have already written better code for generic keyboards.  

At best I, down the road, I’d eventually support two notes at once so that you could add a third single additional note and make triads, but probably not more.

bazmonk, 2022-03-15:
I'm closing this issue to keep things tidy... drone/trompette on/off has been added in to the code, legacy keycombos are there too, I've established what--if anything--I'll do about polyphonic notes, I think we're good with the tenor keybox stuff, and I think that covers Herve's original request.

Feel free to open another issue about something specific if I'm missing something here.

closed 2022-03-15 by bazmonk