bazmonk/digigurdy-baz #30: fatal error

lune36400, 2022-03-09:
today i try to upload on teensy 3.5 but fatal error
[image]

the error is 

Arduino : 1.8.13 (Windows 10), TD: 1.56, Carte : "Teensy 3.5, Serial + MIDI, 120 MHz, Faster, French"

C:\Users\Hervé\AppData\Local\Temp\Temp1_digigurdy-baz-main (2).zip\digigurdy-baz-main\digigurdy-baz\digigurdy-baz.ino:123:46: fatal error: bitmaps.h: No such file or directory

compilation terminated.

Erreur de compilation pour la carte Teensy 3.5

Ce rapport pourrait être plus détaillé avec
l'option "Afficher les résultats détaillés de la compilation"
activée dans Fichier -> Préférences.

bazmonk, 2022-03-09:
There are three extra header files that need to be in the same directory/folder as the `digigurdy-baz.ino` file when you compile:

```
// These are found in the digigurdy-baz repository
#include "bitmaps.h"       // Pretty pictures
#include "eeprom_values.h" // Save-slot memory addresses
#include "default_tunings.h" // Preset tunings.
```

Can you check those files are present?

lune36400, 2022-03-09:
sorry it's ok now,

I think this night i need for my new pcb, because is a combo  dg et hg

no change for the menu and detect crank motor, the soft is ok for me,

-at least,  the dg run directly on play (without push gaming button, stay like that if i want play dg only.

-a button to stop open string (the drone can be set in menu or in bs16i)

-a button to stop when hg play without dg all the time (if all the time i cut dg)
 -
- the 3 button for capo
 -2 button for  x and o

i think i must stop the pcb before all the new soft is design.

i can add and comment and uncomment if you telle me where

(where you come from, fb is unpersonnal, me to center of france near st chartier (festival 3km)

thanks a lot

rv

[image]

bazmonk, 2022-03-09:
> (where you come from, fb is unpersonnal, me to center of france near st chartier (festival 3km)

I'm in California, US.  I don't use facebook much except to connect with digigurdy and disc golf people.