bazmonk/digigurdy-baz #33: Menu options

Gubbledenut, 2022-03-10:
Working through , so please leave this comment live, more to come!!
When I select a preset the go to turn off the drone and/or trompette the screen is cluttered, and the only way to continue appears to be save tuning, as the O button is toggling back the drone/ trompette ???
Think given that we have limited lines needs a little more ergonomic planning..
Also is there a way please that the preset tuning's "titles" as well as designated notes can all be in the default tunings subfile?
While it is very simple to change the preset , having to go back to main code to change it's displayed "title" seems a tad onerous??
More to come on this menu stuff, sorry
David

bazmonk, 2022-03-10:
From the pause menu (where hitting O cycles through the drone/tromp on/off), X or 5 button takes you back to just playing, no need to go through a save or load screen.  I made that the X button in the sense of going "back" to the playing mode, but I agree that pause menu is busy.  I think I will combine loading from preset or save slot to a general "load" screen, so that I can save a precious line and clean up the main pause screen.

> Also is there a way please that the preset tuning's "titles" as well as designated notes can all be in the default tunings subfile?

Yes, can you make a separate issue for this?  If not I'll try to remember to later today.  That's a good idea.

Not sure if you noticed, but the presets aren't quite what John's presets actually are.  One of them was mis-commented from what it actually was so it was a choice between what it looked like it was supposed to be, and what it actually was.

If you tweak the presets and think they are a better set of four to offer there, please show me, because I'm not that familiar with common gurdy tunings specifically.

bazmonk, 2022-03-13:
Handling the preset titles in preset file thing in issue #34

The above commit cleans things up a little.  Presets have a preview screen like the loading a save slot does.  You can back out of all the options without saving/changing tuning now (previously if you choose to pick a new custom tuning, you were stuck until you had to choose one).  I've separated the Go-Back and Drone/Tromp on/off options from the load/save tuning options, hopefully clearing that up a little bit.

Closing this issue.

closed 2022-03-15 by bazmonk