bazmonk/digigurdy-baz #34: Default Tunings

Gubbledenut, 2022-03-10:
As Requested,
Please put the preset titles as well as the notes in the same sub file to make adjustments a bit easier.
Me and Sergio designed and allocated  those presets, so my fault not John's , however the C drone is low and the trompette often C3 too but the G trompette is higher, dunno why.
Ideally the goto tuning ie no 1 should be  the G Gg G one, it's the most versatile for beginners , while in G/C tuning  C Gg C is ore "traditional" it's restricted to only being ab le to play in C, where the all G one lets you play in G or C without having to tune.

designating the tunings in the same sequence they are as you look down it's Trompette, Chanter Low, Chanter Hi, Drone ie C Gg C
just easier to visualise that way..
So keep the same 4 just alter to this perhaps, so the two unison tunings are first in each ??
1;  G Gg G
2; C Gg C
3; D Dd D
4; G Dd G

bazmonk, 2022-03-15:
I'm pushing this past v1.0 for now.

1. Are you saying what I have as the presets aren't the correct presets that you and Sergio designed?  If so, could you (either here or in a branch/pull-request let me know what they should be and the order you'd like them suggested in?
2. Changing the tuning sequence would be a nontrivial change, as there's several menu screens, menu sequences, and displays that use this order.  There's also the MIDI channels themselves... I think these should all match.  I'm down for changing the order, but I'd want to change more than just the presets at the same time.  I sorta like having the reverse of what you propose: Drone, Chanter hi, Chanter low, Trompette.  On the display itself the drone is at the "top" from your perspective looking down at it so this would line up with the layout of a gurdy you're holding versus looking at from the third-person.  Plus, the high chanter is the one pegged to the note being displayed (is _that_ what it should be?), and I think the chanter that corresponds with the displayed note should be listed first just on principle.

Gubbledenut, 2022-03-15:
Happy with current preset choices at ver0991,if anyone needs something more complex then the advanced tuning options along with transpose are fine....but let me think on it, and I'll also discuss it with Sergio.

bazmonk, 2022-03-26:
Finally got around to this one.  See the above commit, v1.1.6.  This builds fine, but I'm packing it in with another change before I call it v1.2 and merge it back into the major `testing` branch.

Each PRESET# needs to have a PRESET#_NAME that goes with it.  Max size is up to you to follow, but I noted it in the `default_tunings.h` comments for reference.

Also flip-flopped choice 1 and 2 per your suggestion.

I'm going to convert this issue to a discussion now that the main request is completed.  I want to keep it around if/when I revisit changing the string display order all-around.