bazmonk/digigurdy-baz #36: Menu display suggestion

Gubbledenut, 2022-03-11:
Have on line along the top with the tuning showing in format C Cc C only as in trompette low hi chanter drone 
Then below large note/ DoReMi 
Could use 0 to designate muted tr or drone..oh and maybe to line transpose and capo status..then shorthand tuning second line ..four lines left to display note ?

bazmonk, 2022-03-15:
> Will try to see it transpose and capo stay will fit eg.
> T-12 C3 C4c5 C2 C+1

With sharps and flats in the mix it won't.  The size 1 font gives you 21-character lines exactly.

We've got more horizontal space to work with.  A third display option could be "Big note on left + compact readout on the right".

bazmonk, 2022-03-15:
Didn't mean to close this :-)

closed 2022-03-15 by bazmonk

Gubbledenut, 2022-03-15:
it's unlikely you will have sharps and flats in tunings..if it's a transpose then live with it, but playing in minor key will still only have full notes tuning
it's another argument to just have the common sharps and flats listed and used for the keys we play in, rather than everything, sure add an option for everything, but for mostly G/C and D/G tunes, and for a learning point perspective, I think it's more sense to use below rather than everything???
So ;
Db is not used as much as C#
D# is not used as much as Eb
Gb is not used as much as F#
A# is not used as much as Bb
Ab is not used as much as G#

bazmonk, 2022-11-03:
Let's see what becomes possible with variable-width fonts.  They're skinnier :-)
`Let's see what becomes possible with variable-width fonts.  They're skinnier :-)`

u8g2 should buy us a little more room.  I'm not super-looking forward to re-doing the notes and staff.  I did some crazy array stuff to align all those stupid bitmaps ;-)

But, maybe a third play screen option (or fourth I guess), that's completely re-imagined with proper spacing of nice fonts and more status info and I can just print the note names in a large font instead of generating bitmaps ever again.

Gubbledenut, 2022-11-11:
ok love the new fonts and setup in the u8g2 test
I like the idea of a rotary push knob, but as John suggests perhaps for most of the venues repurpose the +/- buttons as up/down in menus with A as enter...then out of menus while playing the +/- go back to being transpose ones? 
I can easily put in a rotary/push thing...but as we would still have the  pushbuttons for capo, Transpose +/- and crank ..plus the three Ex ones..may as well utilise them more fully?