bazmonk/digigurdy-baz #39: BS16i Scenes issues

Gubbledenut, 2022-03-12:
Tried to load run your two BS16i plists, they look for sf2 files apart from ViennaAlto.11 which most will not have..so I have odified the plist to remove all but viennaAlto 11..loads fine now 
...I do have the raw Wav files for the French 2 , ViennaSoprano and Keyclicks but not the SF2 files , so if you want to upload those here too that would be wonderfu
[image]
l

bazmonk, 2022-03-12:
Glad to know they at least work in theory!

Those are more soundfonts from the MIDIgurdy project.  Here’s the link to them:

https://github.com/midigurdy/mg-initialdata/tree/master/data/sounds

But what I’ll do is re-make those bs-16i “scenes” with only the mod11 version that John distributes.

bazmonk, 2022-03-15:
Didn't lose track of this, but I'm still experimenting with the scale tuning settings, and I wanna make the re-made version of the plist nicey-nice while I'm at it.

Something I've had a little fun with is a second trompette buzz... you can duplicate MIDI channels to multiple paths in bs-16i, and by tweaking the tuning/effects a bit and using a different buzz sound, there's a richer buzz sound to be had I think.

bazmonk, 2022-03-21:
I’ve put up (testing branch) a bs-16i scene plist that should only look for the Vienna-alto-mod sf2 and the default that comes installed with it.  Let me know if that one works better and I can close this out.

Gubbledenut, 2022-03-21:
That works, thanks, another close off ....

closed 2022-03-21 by bazmonk