bazmonk/digigurdy-baz #43: 0.9.8 Button 6 issue

Gubbledenut, 2022-03-13:
Button/option 6 doesn't seem to be working in the trompette selection screen
Everywhere else it seems fine, and it's playing so not my hardware!
Very much like the new display by the way...

bazmonk, 2022-03-13:
Which screen is that?  The screen where you choose between the six choices when making your own tuning?  That works for me and the code looks right:

```
  while (!done) {

    my1Button->update();
    my2Button->update();
    my3Button->update();
    my4Button->update();
    my5Button->update();
    my6Button->update();
    myOkButton->update();

    if (my1Button->wasPressed() || myOkButton->wasPressed()) {
      mytromp->setOpenNote(choice1);
      done = true;
    } else if (my2Button->wasPressed()) {
      mytromp->setOpenNote(choice2);
      done = true;
    } else if (my3Button->wasPressed()) {
      mytromp->setOpenNote(choice3);
      done = true;
    } else if (my4Button->wasPressed()) {
      mytromp->setOpenNote(choice4);
      done = true;
    } else if (my5Button->wasPressed()) {
      mytromp->setOpenNote(choice5);
      done = true;
    } else if (my6Button->wasPressed()) {
      mytromp->setOpenNote(choice6);
      done = true;
    };
```

Or do you mean on the pause screen where you can turn the drone/trompette on/off?  That I only linked to the O button on purpose and not also a number, but I can...

Gubbledenut, 2022-03-14:
I load your unmodified 0.9.9  It will let me select option 6 for the Drone and trompette 
But does not seem to allow me now to Select 1 for the display option...2 is fine

The option 6 and option 1 display as above do not seem to work in my extended key map...not a biggie so no rush everything else works, and oddly the option 1 works fine in the tuning screens...my key array below.
/// NOTE: There's no key that produces 0 offset (an "open" string),
// so the first element is bogus.  It gets skipped entirely.
/*int pin_array[] = {-1, 2, 24, 3, 25, 26, 4, 27, 5, 28, 29, 6, 30,
                   7, 31, 8, 32, 33, 18, 34, 19, 35, 36, 20, 37};*/

       //DJ Array to include Tenor pins for 4.1
 int pin_array[] = {-1, 54, 53, 52, 51, 50, 49, 48, 2, 24, 3, 25, 26, 4, 27, 5, 28, 29, 6, 30,
                   7, 31, 8, 32, 33, 18, 34, 19, 35, 36, 20, 37};            

// This is literally just the size of the above array minus one.  I need this as a const to
// declare the KeyboxButton array later on... or I just don't know enough C++ to know how to
// *not* need it ;-)
//const int num_keys = 24;
//DJ Tenor keybaord for 4.1
const int num_keys = 31;

bazmonk, 2022-03-14:
I'm wondering if your problem is where I assign the 1-6 buttons:

```
  // 1-6 keys are the first six bottom keys of the keybox
  my1Button = mygurdy->keybox[1];
  my2Button = mygurdy->keybox[3];
  my3Button = mygurdy->keybox[4];
  my4Button = mygurdy->keybox[6];
  my5Button = mygurdy->keybox[8];
  my6Button = mygurdy->keybox[9];
```
If your keybox starts with a different pattern of sharps and natural notes than a standard keybox, these won't line up with the bottom row keys.

Gubbledenut, 2022-03-14:
That makes sense, so what do the numbers in the [ ] refer to relative to the config higher up...?
happy to adjust stuff and test..no rush

bazmonk, 2022-03-14:
The numbers are the position of the keys on the keybox, minus one, in chromatic order.

On a normal keybox tuned to G, the very first button (G#) is keybox[0].  keybox[1] is the first bottom key: A.  Keybox[2] is A#, and then keybox[3] is the second bottom key, B.  Etc. etc.

If I'm looking at the pic you sent me of your keybox once, I think my6button needs to be [10].  If I'm right, that means when your button #6 wasn't working, the flat just below it was the one misconfigured as the 6button.

Gubbledenut, 2022-03-14:
yes that works...I'll add 
my6Button = mygurdy->keybox[10]; //DJ Added
to my code...I guess the next question is can the mybutton defines be included with the rest of the config items at the start of the code?

bazmonk, 2022-03-14:
Yes... I'll make const variables you can set in the config section to adjust these.

bazmonk, 2022-03-15:
There ya go.  I'm gonna close this issue.

closed 2022-03-15 by bazmonk