bazmonk/digigurdy-baz #49: Coup not possible

Gubbledenut, 2022-03-15:
ver0991 the buzz seems to come on but it's not controllable as in being able to perform a coup correctly.
I have tried with setting from 7 to 60 and it does slightly alter the sensitivity but I cannot control it to obtain a coup/ beat??
at least I have the buzz working now.

Gubbledenut, 2022-03-15:
I’ll do some more testing, including trying the original codes with a 15 and 25 thresholds….yes I’m using a slightly different motor, I did try the cheaper Chinese ones, but went instead with polou branded ones, and they also have slightly different gearing…but that’s not been a big issue up.till now….I’ll let you know..thanks for moving buttons to config section by the way.

bazmonk, 2022-03-15:
I don't have the code in front of me, but IIRC, the v_threshold of 25 only pertains to when the cranking starts sounding, not when the buzz starts.  For buzzing itself, that shouldn't affect it unless the voltage coming out is too low and the crank is coming on/off intermittently while you're cranking it.

Gubbledenut, 2022-03-15:
Let me try a few things out... John's emailed suggesting it might just be a particular sample...leave this open until I can narrow it down. Spent much of my working life trying to break  high end electronic things before they went out to the customers.... Was called test engineering then 😎

Gubbledenut, 2022-03-18:
I have it working now, changed the  V threshold to 15 (with the resistors it's running lower voltage) and the loop delay to 60 (running a 4.1) still may tweak the buzz smoothing and decay to see differences...but it is now working but think it can work better....would there be a correlation between the voltage range produced and the signal into pin 15??? For John's made ones with the same motors and resistors, I imagine the voltage range will be pretty consistent....looking for max I imagine as the lower limit will be negligible up to 2 or 2.5v ish??? ok so happy now addressed  this issue given correct parameters ...

bazmonk, 2022-03-18:
Glad to hear the v_threshold did the trick for you!

> would there be a correlation between the voltage range produced and the signal into pin 15??? For John's made ones with the same motors and resistors, I imagine the voltage range will be pretty consistent....looking for max I imagine as the lower limit will be negligible up to 2 or 2.5v ish???

Sorry, I'm not quite sure what you mean here...

The `analogRead(pin)` function returns a range by default between 0 and 1023, with 0 being "ground" and 1023 being 3.3v in this case (I think, given what I read after our last convo on facebook).  

So the v_threshold we're going with, even at 25, is very low, and I think the goal here is to make it as low as possible without random fluctuations in the voltage causing it to register, so that even slow movement triggers it.  I left mine at 25 because it worked fine for me, and I didn't yet appreciate the variance different motors give :-)
  
I won't muck with the main code branch until after v1.0, but I'm looking in to using the ADC library to do this in a smarter way than just pure `analogRead()` calls.  There are ways to do continuous reads that buffer the value for you to read, supposedly more rapidly than waiting for the actual hardware polling to happen, and ways to do differential reads between two pins (e.g. compare the buzz potentiometer pin directly to the crank pin) which ensure the two reads are synchronized, which should cut out some of the wandering voltage values due to the actual charge of the "ground" fluctuating (they'll both supposedly fluctuate together instead).

Doing experiments right now to actually time how long stuff is taking here, if other methods can actually read it more smoothly or more quickly, etc.  If any of that works, in theory a lot of this can be made more sensitive (faster buzz response without randomly kicking in, more responsive crank).

And I'm not sure, but *maybe* if it can be made smooth enough somehow (like a continuous average), there could be a "calibration" option in the code, where you're asked to crank at your full speed, and it determines how high your crank runs and normalizes the raw voltage readings to behave about the same no matter what.  It would set things like a good v_threshold for you.

bazmonk, 2022-03-18:
@Gubbledenut WOWOW:

This does 10,000 analogRead() in about 75ms:

```
int v_pin;

int interval = 10000;
int v_arr[10000];

void setup() {
  // put your setup code here, to run once:
  v_pin = A1;
  pinMode(v_pin, INPUT);
  Serial.begin(38400);
  delay(3000);
  Serial.println("Hello.");
}

void loop() {
  // put your main code here, to run repeatedly:
  int start_time = millis();
  int stop_time;
  for (int x = 0; x < interval; x++) {
    v_arr[x] = analogRead(v_pin);
  };
  stop_time = millis();

  int total = 0;
  for (int x = 0; x < interval; x++) {
    total += v_arr[x];
  };

  Serial.print("Elapsed time for ");
  Serial.print(interval);
  Serial.print(" reads (ms): ");
  Serial.print(stop_time - start_time);
  Serial.print(".  Average voltage: ");
  Serial.println((float)(total) / interval);
}
```

This does it in LESS THAN 1ms!!!

```
#include <ADC.h>

int v_pin;

int interval = 10000;
int v_arr[10000];
ADC *myadc = new ADC();

void setup() {
  // put your setup code here, to run once:
  v_pin = A1;
  pinMode(v_pin, INPUT);
  myadc->adc0->setConversionSpeed(ADC_CONVERSION_SPEED::HIGH_SPEED);
  myadc->adc0->startContinuous(v_pin);
  Serial.begin(38400);
  delay(3000);
  Serial.println("Hello.");
}

void loop() {
  // put your main code here, to run repeatedly:
  int start_time = millis();
  int stop_time;
  for (int x = 0; x < interval; x++) {
    v_arr[x] = myadc->adc0->analogReadContinuous();
  };
  stop_time = millis();

  int total = 0;
  for (int x = 0; x < interval; x++) {
    total += v_arr[x];
  };

  Serial.print("Elapsed time for ");
  Serial.print(interval);
  Serial.print(" reads (ms): ");
  Serial.print(stop_time - start_time);
  Serial.print(".  Average voltage: ");
  Serial.println((float)(total) / interval);
}
```

We're cookin' with gas!!!

bazmonk, 2022-03-19:
I'm starting a testing branch that I'll be adding this into.  In the "continuous_crank" I already have an almost working (crank detection broken) version that uses continuous read.

It works GOOD.  Before the digigurdy code was doing 16 analogreads per 1ms.  Now it's doing **11,000**.  I'm actually taking an average of 4,000 reads at a time, ending up with about 2.5 "reads" per 1ms.  But they are super-consistent averages instead of wandering values and do not wander, and eliminating the analogreads has allowed me to remove the LOOP_DELAY again.  

The consistency of the reads means you/I can set those buzz and spin thresholds very low now and not worry about phantom noises creeping in.  You should be able to fine-tune the buzz to be as sensitive as you want it to be for your coups.

I'll probably have a version you can play with on github tonight.  I think you'll like it!

Gubbledenut, 2022-03-20:
Been playing with the latest test version, continuous crank, I’ve left the settings as is, using your/my scenes in bs16i and it’s very playable as in im playing tunes not just testing functionality! The buzz/tirant feel is as good as the real things…it’s very very good.
I’ve put all my controls next to my screen, including a micoswitch for the crank/mute…very easy to control, and see. Oh yea and it’s on my extended 31 keybox…think this is a real winner..and leave the adjustments in the code, maybe in the config section? Not as an option in the interface? Over complications kinda ruin the effect?

bazmonk, 2022-03-20:
Yeah, I probably won’t add adjustments to the menus for this stuff.

I tested yesterday on my 4.1 unit and as I suspected, increasing the number of samples per cycle so that it takes about the same time as the 3.5 does per cycle (I’m aiming for about 40ms per 100 cycles, but that’s just trial and error) makes it “feel” the same as far as I can tell.

So something I might do: part of the crank detection at the beginning can be some code that briefly measures how long it’s taking and adjust to take 40ms.  So the code could possibly “calibrate” itself for the CPU speed upon startup.

Even that, though, is a low-priority feature, because it’s not like you can switch CPUs without also switching code, and changing “4000” to “11000” before compiling isn’t a daunting task :-)

I’ve only ever touched/played my nerdy gurdy, so it’s great to hear from someone that has more experience with real gurdies that it feels good.

bazmonk, 2022-03-21:
Just tidying up and closing this issue.  I think the continuous analog reading fixed the core issue here and the other timing variables give enough flexibility to adjust the responsiveness.

closed 2022-03-21 by bazmonk