bazmonk/digigurdy-baz #56: Feature: Crank redetection?

bazmonk, 2022-03-21:
### Discussed in https://github.com/bazmonk/digigurdy-baz/discussions/48

<div type='discussions-op-text'>

<sup>Originally posted by **bazmonk** March  1, 2022</sup>
Right now the setup() includes a step that samples the crank voltage for a few seconds, and calculated how much that voltage is wandering.  An unconnected pin’s voltage will wander significantly more than the crank plugged in but at rest, and this is how digigurdy detects its crank at startup.

It’d be cool to find a strategy for 1) gracefully recognizing when the crank becomes unplugged, and 2) periodically check when the crank isn’t plugged in to see if it’s there now.  In other words, support docking and un-docking with the crank on-the-fly.

I’m not sure how to periodically sample the crank without interrupting play.  It’d have to be built into the loop() (i.e. sampling as it goes, not the loop with delay() that we have now).

A compromise that shouldn’t be as bad is a menu option to just tell the digigurdy to re-detect or disengage the crank when you want to dock/undock it.

No milestone for this one… still mulling over it.  Right now the crank is annoying me in the rewrite code :-)</div>

Gubbledenut, 2022-03-24:
Can close this, maybe put note in code "if crank isn't detected swap wires on rectifier output to gnd and pin 15 around "

bazmonk, 2022-03-24:
Oooh, is your teensy ok?  I don’t know how much it takes but I know it doesn’t like negative voltage…

Gubbledenut, 2022-03-24:
yea it's fine so far, but I will check the voltage again. Crank action is a bit jittery , and doesn't like low speeds, adjusting the parameters and compiling to see what makes a difference, running the 4.1 will adjust threshold a bit more tried 5000 and 15000, but might try few others.

bazmonk, 2022-03-24:
I’m worried that the poor performance at low speeds is because of the minimum voltage necessary to get across the diodes.  If that’s the case, none of the parameters would help because there’s just no voltage to measure at low speed and it really is that jittery, you know?

Gubbledenut, 2022-03-24:
Ok that makes sense, that it's not sensing🤓. I may then go back to no rectifier on my player, but do some more experimenting on my test rig with another motor.
The bidirectional cranking is used in getting a staccato effect and very slow revs in playing baroque gurdy ...look out YouTube tobie miller and you'll see what I mean...

bazmonk, 2022-03-26:
https://github.com/bazmonk/digigurdy-baz/pull/58

I'm closing this with the addition of this option to re-detect the crank during play.  Does the job.  I can revisit auto-detection later...

closed 2022-03-26 by bazmonk