bazmonk/digigurdy-baz #59: screen ver 1.2

lune36400, 2022-03-27:
Je viens de regarder ton nouvel code 1.2, 

quand on appuis avec le bouton "gaming" l'affichage de l'ecran, est ok

par contre 

en tournant la manivelle en arriere, et doucement avec l'ancienne version moteur, l'ecran affiche "la note joué" puis l'ecran fixe ou il y a  les "données de selection"

en tournant dans le bon sens l'ecran des notes jouées s'affiche normalement , mais le moindre rythme un peu plus lent, repasse sur l"ecran de "données selection",  et parfois quand la vitesse est lente, les deux (note et selection  s'affiche en meme temps, je pense que cet ecran de selecrion devrait etre temporisé de quelques seconde, pour eviter  ce cumul, et d'avoir ce desagrement.

j'espere avoir eté clair???

bazmonk, 2022-03-27:
Using Google Translate :-)

> By turning the crank back, and gently with the old engine version, the screen displays "the note played" then the fixed screen where there are the "selection data"

> By turning in the right direction the screen of the notes played is displayed normally, but the slightest pace a little slower, goes back to the "selection data" screen, and sometimes when the speed is slow, both (note and selection are displayed at the same time

I am not sure if I understand you correctly.  Do you mean that the crank is jittery and only plays very briefly?  It doesn’t stay on consistently at slow cranking speed?  I think that’s what you mean by the note and selection screen coming on “at the same time”.

The motor doesn’t behave equally in both directions, so I would expect it’s more noticeable in one direction than the other.  Am I understanding your question correctly?

You’re using a diode bridge that John mentioned, correct?

lune36400, 2022-03-27:
for the moment i have no diode  bridge, i  use the old motor version.  in first i try the crank in return way very slowly, the display coming and have two screen "note" and after "selection".
  and when i crank slowly the display see the "note", and when we have little stop of crank the "selection" come immediatly, i thinks it's necessary to delay a litlle the "selection"   and  "note"  stay a litle more delay or blank the display.

it's difficult for me to explain rigtly....

bazmonk, 2022-03-27:
When this happens, do you also hear the note playing come on and off quickly?  It sounds like it’s not a problem with the display, but a problem with the settings betting too sensitive for your crank.

If you hear the sound stuttering as well as the screen, then it’s the crank sensitivity.  You can try:

* Raising the `V_THRESHOLD` value to a higher one.  I use 8, which is rather low.  15-25 was the original range John’s code used.  This will make the crank less sensitive at low speeds.  Ideally you want to set this so that the crank doesn’t kick in until you’re spinning it at the lowest speed you’d want to play it, so that instead of “stuttering” out at lower crank speed it simply cuts off.

* Increase the `MAX_SPIN` and `SPIN_WEIGHT` a bit.  This will make the note stay on longer when the crank voltage momentarily drops, so it will play through the “stuttering” events.  The downside is this also adds a delay to how quickly the sound will stop when you stop cranking, so you only want these values to be high enough to be smooth for your playing, and not any higher than that.

* You may also need to raise the `BUZZ_SMOOTHING` number if that is also too sensitive.

bazmonk, 2022-11-03:
I'm going to close this, because when I re-implement the motor crank (which is my next goal after bringing the Blue OLED back into play, like v2.3), I'll be redesigning it from the ground up.  Same idea, but with a smarter use of timers and hopefully your issue won't come into play there.

closed 2022-11-03 by bazmonk