bazmonk/digigurdy-baz #62: Needs an "about" screen

Gubbledenut, 2022-03-28:
suggest under the other options menu as option 2 for remove crank is option 1?
Needs to have the version number easily getable in case of any issues...and especially with the frequency of updates you're posting..I'm struggling to keep up today ;-) Awesome work thank you. I just need to find a better sample from a C1 drone

bazmonk, 2022-03-28:
Thanks for making an issue!

What do you think: should I just announce changes less often?  I’m gonna make ‘em as fast as I make ‘em, but I can just make weekly updates during times I’m working faster than that.

Gubbledenut, 2022-03-28:
The fact so few digigurdy owners can or will upgrade their own instruments, is a big factor. I am more than happy to upgrade every few hours. I suggest aiming for a major upgrade release call it perhaps 1.5 , as we know ver 1.0 is pretty good.
Then suggest make that the next one we offer to upgrade everyone's processors if they want it. By then I think we will have ironed out any ambiguities or bits that might not be as intuitive as they can be, again we are talking about the majority of digigurdy users not the four or five who are willing to upgrade the units themselves?

bazmonk, 2022-03-29:
I forgot to reference in the commit, but this about screen has been implemented.

I'm going to try for just less-frequent update announcements.  I agree many users don't/won't want to update their unit that frequently, but I have a few reservations about only announcing "major" upgrades:

1. I'm announcing it _for_ people like you that would be more inclined to try it out.  I figured other users can just wait until a new feature piques their interest, or I declare it a "major" version.
2. I have no idea what/when major releases are :-). It's pretty arbitrary.  The changes to the crank I made just after v1.0... IMO that's worth upgrading for by itself, but I don't want to turn around and just declare a 1.5 or 2.0 that quickly.  It will slow down... I really am running out of big improvements, so in the not-too-distant future I'll be adding stuff less frequently.
3.  Updating the code on the Teensy is very easy in the grand scheme of things.  I hope I do coax some skiddish people into trying it themselves.  It's less effort than opening up the gurdy, removing the unit, shipping it, and waiting... for everyone involved.

I'm new to facebook and not sure how that "featured" section on the group page works, but maybe John will let me use one of those spots to hold some info on the latest version and changes, and I can update that freely without clogging up the discussion feed.

Gubbledenut, 2022-03-29:
Meh, clutter up the Facebook feed, I'm happy to follow things here and download , try to break anything! I like the different branches, so keeping the testing and main , also agree up to you when you want to push anything to general release. And what you call it. True updating the teensy isn't difficult, assuming everything is packaged into a single folder along with main code and sub files, named the current version rather than just "....baz master" . The main issue with uploading is the contradiction of what to use, the teensy software is what I use rather than from arduino download site...then it's important the right libraries are easily accessible and also downloaded to avoid compile errors...which are not always obvious where the errors are. Given previous experience it is far less hassle to offer to upgrade any hesistant users chips, you won't be the only one who can do it..few people in the US and others worldwide..just need to send it to one of use with a postage paid return address pack included....of course we need to check which screen they have..but I imaging john has those details.
The digigurdy is a plug and play, with now even cooler possibilities...it is a little unfair to assume everyone wants to delve into recoding theirs ....

bazmonk, 2022-03-29:
I did look into distributing the code in compiled form, so that they wouldn't have to worry about the libraries or setting up the teensy environment just-right, and could simply load it up... but I abandoned the idea because:

* The teensy loader is packaged in the teensyduino stuff and it isn't straightforward to get just the loader.
* Loading a HEX file directly w/o compiling it through teensyduino requires pressing the little button on the unit to put it in programming mode.  As far as I can tell the "trick" that the IDE does to put it in that mode via software isn't easily available.

So it ends up being tricky in its own way and not really any easier.

You're right: updating for people isn't a big deal.  I really don't have a good idea of how many digigurdies even exist or how often people approach John (or you?  I'm confused if there's an arrangement here I'm not aware of or if this is just one big happy help-where-you-can thing) for updates.   I see 630 members of the group and I hear from like, 5 of us :-)

Gubbledenut, 2022-03-29:
You need to email John, no one approaches me ..which I prefer. I have from the very start of the project done my own thing, testing john's and now your code..much like Herve (lune) does, but he's also a rather accomplished luthier...He's making his projects in conjunction with Greg Jolivet ( gurdy legend)who's a neighbour. 
I have tried to get more people involved on the Facebook page, I think very few have made their own from scratch, so again John has details of the ones he's sold along with the versions.

bazmonk, 2022-04-04:
> True updating the teensy isn't difficult, assuming everything is packaged into a single folder along with main code and sub files, named the current version rather than just "....baz master" . 

Forgot to mention this, but for what it's worth, that's if you grab a specific branch copy off the site.  I realize that's what most people would do, but in your situation (somewhat frequently trying out a new version of my code), you should clone the entire repository and use git directly.  That would give you full local access to all the branches like I have on my laptop, and you'd be able to easily pull the latest changes with a click, switch back and forth between different branches or commits (to back out of a new update and grab an older copy, for example).  And if you did it that way, the directory would just be called "digigurdy-baz" like the sketch file and it'd work "out-of-the-box".

If you ever got to having your own versions of the code for your own custom applications or whatever, we could set you up here with your own branch(es) within the repository here that you could merge my future changes into as you please, and modify as you please.  You can do all this locally already, but if they're modifications that would be useful to others (tenor keyboxes or more buttons for your control panel or whatever), we can keep them here, too.

Gubbledenut, 2022-04-04:
I have cloned it already, and then adjust to add the extra keys and changed button 6 position ,
maybe once there is a major version change I will add a tenor keybox even a harpa one maybe..but for now to avoid confusion think stick with just you contributing to the branches as you go.

bazmonk, 2022-04-15:
Closing this: we’ve got an about screen now.

closed 2022-04-15 by bazmonk