bazmonk/digigurdy-baz #65: new idea

lune36400, 2022-04-03:
Hi,

This night  a  new idea.

-possibility to change on fly with little delay, or not ( as it's possible)  to change one slot of the 4 previous slot save before ( possible or not)

in my project i see the possibilty to add note, thanks Basil,  can you add just a button  to have   "on open string" and  "off open string" in hard by comment or uncomment or "on fly" by button alwaws in choise ( the better for me), sorry to have some need,  i can do lot of thing, 

Sorry (David), or Gubbledenut,  i am not a professionnal luthier, i am a noob in this way, i do lot of things for my pleasure, old car, built house, workshop, cnc, 3d printer, i prefer to built than use, i am poor man to be a creator, but i can help for other tings.......i do my hurdy gurdy with the help by Seb Tourny  Luthier in La Chatre 36400.

do you thinks if we use  male pogo pin for connect under teensy,  good idea or not, in this case easy to connect, and easy because removable, no soldering.....

like this  
[image]

Gubbledenut, 2022-04-03:
I put all my Teensy into sockets  onto boards  , I don't solder directly to any of the teensy so they can be swapped...except for the tenor extra keys I do need to solder directly onto teensy ...
Your "open" string note is on a teensy pin ? but it is not using one of the keys itself???
So in the code  from line 75 you  need to add the extra pin after the -1 
Below is the code from line 75 I think ...

//THIS IS THE STANDARD 24 key box
// NOTE: There's no key that produces 0 offset (an "open" string),
// so the first element is bogus.  It gets skipped entirely.
/*int pin_array[] = {-1, 2, 24, 3, 25, 26, 4, 27, 5, 28, 29, 6, 30,
                   7, 31, 8, 32, 33, 18, 34, 19, 35, 36, 20, 37}; */

//TO ADD A G SOL midi 67 on pin 15 
  int pin_array[] = {-1, 15, 2, 24, 3, 25, 26, 4, 27, 5, 28, 29, 6, 30,
                   7, 31, 8, 32, 33, 18, 34, 19, 35, 36, 20, 37};            

// This is literally just the size of the above array minus one.  I need this as a const to
// declare the KeyboxButton array later on... or I just don't know enough C++ to know how to
// *not* need it ;-)
//const int num_keys = 24;
const int num_keys = 25; //YOU NEED TO ADJUST THIS  TO THE NEW TOTAL NUMBER OF KEYS ON THE KEYBOX

bazmonk, 2022-04-03:
Pogo pins... do you mean for the underside surface-mount pin spots?  Maybe... I don't think it'd work on the through-hole pins, but maybe the surface-mount ones.

---

Quick save-slot access: perhaps another hidden keycombo on the keybox could be hot-keys to load slot 1-4.  What is the situation you'd want to have this?  Without another button I don't think it could be made on-the-fly while actually playing, but a keycombo would make it a little faster to access... but I'm not sure if that's immediate enough to achieve what you want.

---

Sorry, I'm not sure what you mean by having an open-note string beyond what @Gubbledenut mentioned for adding to the keybox and what we've talked about before (using the -1 spot in the array).  Does that not work for this situation?

lune36400, 2022-04-03:
pogo it's just for lower pins 40-50, where there is no hole it would be to avoid soldering directly on the teensy, and to be able to use it as a test unit, it's always ideas to try.

indeed the possibility of being able to change range, therefore slot, 
  a good player could change by easily accessible additional buttons, one of the 4 slots quick save already configured by playing in fly. it 's for the good players, like greg (he is very intresting about it) 

Thank you for the possible modification at the level of the open string by adding a 25th key, but it is to be able without changing the shcetch inside the teensy the possibility of using as the dg therefore with the open string (2 possibility modified by a button).

it's going to be a lot of buttons, but a lot of game possibilities

bazmonk, 2022-04-03:
This page describes how you can use a little chip controller to effectively put more than one button on a pin.

https://www.pjrc.com/teensy/td_midi.html

Just an idea if you're running low on pins.  I'd be interested to know if pogos work out for you, though.

---

I probably won't go as far as making the open-string-button modification part of the main code as an option.  There's two big reasons:

* I don't have one.  It's very hard to write in functionality that I can't test.
* It's a significant departure from the "normal" digigurdy and that's still the target audience of the main code.

The best I can do here is show you how where to modify the code to do what you want, as I did with the open-string-button.  I'd suggest maintaining a fork of the code.  That way you can have these modifications in-place, and when I release updates, you can pull that and merge in your modifications.  I won't be modifying the keybox portion of the code often, so most of the time this merge will be automatic.

---

Quick-change save slots: it sounds like a keybox-combo wouldn't achieve the on-the-fly changing you want (it'd require hitting random notes during play to do it).  So to do with with an extra button, like with the open-note-string, you'd have to do this yourself.  It's not an option I'd include because it's useless on the 99.9% of digigurdies that won't have this extra button... including mine :-)

That said, I did make it easy for you to add this.  Loading a save slot is very easy:

```
load_saved_tunings(slot);
```

`slot` is one of the four EEPROM_SLOT[1-4] values.  You'd just need to initialize a GurdyButton object for it where I initialize the tpose/capo buttons, and then in the main loop where I update and check if the other buttons are pressed, update yours and include something like:

```
if(mySaveButton->wasPressed()) {
   load_saved_tunings(slot);
}
```

(you'd have to do more to determine which slot or cycle though them, but this is the basic idea)

I'd be happy to help you get it working, but I won't be including this in the main code is all.

lune36400, 2022-04-03:
thanks for all,  i have already idea, and lot of time i can don't do it because i have not enough science of it.

this project is alwaws on long time, now i have time, and lot of other hobby, and now the good weather come, more time...

bazmonk, 2022-04-03:
Would it help if I showed you an example (not in the main code, but a
version you can work with) that has an extra button and cycles through the
save slots?  Then you could just copy the idea for more buttons, or modify
exactly how it behaves.

lune36400, 2022-04-03:
for sure, it can help me after i try it in my sketch

for the moment i change the design of the pcb to add  all pin Under teensy 3.5 for pin 40 to 57 and use after the pogo system

Gubbledenut, 2022-04-03:
Herve I suggest soldering wires to the board for the extra internal pins but bring that out onto pins or a connector so it is still removable. This is how I do mine so everything can be swapped out.

bazmonk, 2022-04-04:
Here ya go.  This link shows the changes I added (you can also search for "LUNE" in the code itself):

https://github.com/bazmonk/digigurdy-baz/commit/2432ecaee272ef55dc2967ea76d37ea3ee54f94b

And here's the branch that you can pull/download:

https://github.com/bazmonk/digigurdy-baz/tree/lune-button

This would make a button change to the next slot (starting at 1 when you start up) each time it's pressed.  I marked all the spots where you'd have to add in parts to use a new button.