bazmonk/digigurdy-baz #71: MIDI over Serial ..little confused on declaration in code

Gubbledenut, 2022-04-13:
Just putting serial midi ability into another of my projects and noticed an anomaly in the digigurdy code.

you call these two lines as an example
 usbMIDI.sendNoteOn(note_being_played, midi_volume, midi_channel);
this is one you appear to have declared, as you say teensy doesn't recognise MIDI...it does see below
      MIDI_obj->sendNoteOn(note_being_played, midi_volume, midi_channel);

This is how I added the serial command to John's code so we could use a 5 pin midi and BT dongle, it works ????
      MIDI.sendNoteOn(note_being_played, midi_volume, midi_channel);

bazmonk, 2022-04-13:
I ran into that issue trying to put the sendNoteOn call in an object class (GurdyString).  

Are you calling it from the loop()?

Gubbledenut, 2022-04-13:
ok, leave this up for a while as I ponder it in other scripts..to see mainly if I can get a serial midi output and run a BT dongle....by the way realise there is no easy way to have a USB and 3.5mm headphone /speaker out of any new iPad ....perhaps a hub , or maybe a simple usb hub with an extra usb to 3.5 audio??

bazmonk, 2022-04-13:
I use a little adapter like this with the iPad.  Works great.  I also have an electric guitar interface that accomplishes the same thing, but this is straight to the point.

https://www.target.com/p/heyday-8482-3-5mm-dongle-usb-c-black-gold/-/A-79896868#lnk=sametab

I also tried a low-latency BT dongle meant for gaming.  While the latency was fine, the audio quality was awful.  I ended up returning it.  At the end of the day I'd rather have decent sound quality than a fully wireless setup.

Gubbledenut, 2022-04-13:
Exactly what I looking for, love my wires too! Pity Apple don’t bother making one though. Might be worth putting our various set ups on a Facebook thread.

bazmonk, 2022-04-16:
> to see mainly if I can get a serial midi output and run a BT dongle

What kind of bluetooth dongle?  I'm confused what you're trying to do exactly.

bazmonk, 2022-11-03:
I'm closing this, but check out what the new code does, especially because I've broken up the code into files and I think I'm using the MIDI object more correctly.

The USB MIDI needs nothing. It just exists by magic when you enable Serial+MIDI in teensyduino.  In both the main and gurdystring file, I can just grab `usbMIDI` and start using it.

The Serial MIDI I make the "right" way with `MIDI_CREATE_DEFAULT_INSTANCE()`, which defaults to Serial1, which is the pin 1 tx we use.  That creates the `MIDI` object, which I now use as an object (with MIDI.send...).  Before I was creating it in a funky way and passing it around by pointer (and having to use MIDI->send...) instead.  That was wrong... well not wrong, but unnecessary and cumbersome.

To pass the Serial `MIDI` object (for example to the gurdystring.cpp file), in that file's header (gurdystring.h) I have this:
`extern MIDI_NAMESPACE::MidiInterface<MIDI_NAMESPACE::SerialMIDI<HardwareSerial>> MIDI;`

The `extern` declares that `MIDI` exists somewhere else.  The compiler sort of trusts that in some other file that will call this one, an object exactly named `MIDI` is already there.

I'm still learning a lot about C++, but I think this is more along the lines of a reasonable way to do it.  And it explains the "anomaly".  Closing this issue (cleaning house!)

closed 2022-11-03 by bazmonk