bazmonk/digigurdy-baz #72: Recording

angeleusebio, 2022-04-18:
I think it would be useful for learning, to be able to record songs playing, to later accompany them with the crank while they play, or accompany playing with a string.

bazmonk, 2022-04-18:
If I added this in it would be a far-off feature.

- There’s no persistent memory to work with.  EEPROM is not meant to be used too often and has limited read/write cycles and even if it didn’t, it’s not enough space: 4kb total!  I could hold some in memory, but they couldn’t be saved.  
- I don’t know how to record songs :-). The logical thing would be to record it as MIDI… I’d have to learn to do that, if that’s something that can be done on a teensy in real-ish time.
- In my opinion recording makes more sense on the synthesizer-end (whatever’s powering your speakers), where you have control over what you attach it to and much better software and hardware to work with (a tablet or computer).

angeleusebio, 2022-04-18:
I agree with you, what you say makes more sense than my proposal.  can we close this thread.