bazmonk/digigurdy-baz #75: Volume control

Gubbledenut, 2022-04-27:
Works fine but perhaps instead of using the 0-127 nomination.can we have proportional 0-10 or 11 instead? Think it will be more user friendly...also will be more relevant to the slider settings on bs16i . I am okay Ng again with the RPi thing..and also will make the upcoming tsunami easier to set?

Gubbledenut, 2022-04-27:
0 to 11 will work if you can  increment from 0 silent  in 12.7 steps to 11 =127?

Gubbledenut, 2022-05-02:
ok forget the 0-11 thing, but perhaps give an idea in the load screen of the string notes and their volumes?
ie G3/70 ???G2/30 Once we utilise the tsunami or a Rpi this will  be very useful I think....

bazmonk, 2022-05-02:
Late to this game... if MIDI supports 128 volumes, I want to be able to set them.  So I kinda want to keep the 0-127 selection.

If a user mistakenly thinks it only goes up to 100 and treats it that way, it works fine.

For the second idea about displaying it somewhere, yes I would like to display the volume somewhere.  The issue is space and making it too busy, but I agree displaying the volume somewhere outside the volume control menu is smart.

Gubbledenut, 2022-05-02:
Yes, perhaps add this to the ergonomic design discussion..as mentioned in the big button "issue"

bazmonk, 2022-11-03:
I'm moving this to 2.2:

- I'm going to leave the volume scale true to the number it's actually setting for people really seriously using it as a MIDI controller.  I'm dead-set on "that's how MIDI is and folks just need to know that" :-)
- You do, however, raise a good point with the Trigger/Tsunami volume.  Right now it doesn't honor volume at all.  So that's something to address somehow if possible.
- Trigger/Tsunami understands actual gain values between -70 to +10 for tracks (-70 to +4 for master gain).  I'll probably "translate" that on the volume screen and you'll still be scrolling through the MIDI numbers, but you'll see both adjust.  Something like that.
- One issue is that I'll have trouble stopping the user from selecting gain values that cause a lot of nasty clipping in the audio.  MIDI is all controllable later on so it's not an issue but the Trigger/Tsunami won't stop you.  I think that's going to have to be up to the user, because it depends on what they're doing after my control (like using the onboard amp, or different wavs).