bazmonk/digigurdy-baz #76: Bigbutton

Gubbledenut, 2022-04-27:
I know it is a big button but perhaps it's time to call it something more relevant like crank toggle ? I know there's already a mycrank so maybe mycranktoggle or mycrankswitch ?

Gubbledenut, 2022-05-02:
Sorry one more
for consistency rename mystring and mylowstring to myhistring or myhichant and mylostring or mylochant ????
this will then make them consistent with what you're calling them in the default tunings file???

bazmonk, 2022-05-02:
I'll probably go through at some point and do a larger refactoring of the code and change the "my" names to something more professional sounding.  Will include this then.

I'm setting it to the 2.0 milestone because I want to limit what I add before making a v1.5 that I push to the main branch (v1.0 is not worth using anymore).  I'll refactor after that probably.

Gubbledenut, 2022-05-02:
Once you're at a point when you'd like a hand with it,let me know, spent a lot of time working on the ergonomics of systems, tech manuals and instructions of things I've and others have built. Close these issues and perhaps add to a discussion?

bazmonk, 2022-11-03:
Coming back to this... now that the code is split among lots of files, the "my" thing has kind of standardized into objects I create in the code, and it lets me over-use simple terms like "crank", for example.  Probably won't change that.

I dunno.  I sorta like bigbutton, sorry.  But `myBigButton` to keep with the scheme established.  Which would require changing others.... I'm so on the fence.

2.2, kick the can!  I'll keep chewing on making something that's both consistent, but that I also like when coding.