bazmonk/digigurdy-baz #81: I want to support alternative tunings

SalusaSecondus, 2022-07-20:
(Similar to my past issues, I expect that I'll develop this myself and so am putting an issue here to track it and let people yell at me 😁 )

I play with various instruments which are not modern western instruments and thus do not necessarily conform to A-440 equal temperament. I want to be able to:
1. Change A to be a pitch other than 440. (Either by menu or specific selection.)
2. Change temperament to something other than equal
3. Support non-western tunings (specifically [Arabic Maqam](https://en.wikipedia.org/wiki/Arabic_maqam))

Ideally these properties will all be savable on a per-slot basis, but this needs to be investigated. (Especially since we only have limited storage.)  We still need to figure out how to indicate these to the midi controller, how to safe them, and how much space we need.

bazmonk, 2022-07-21:
I am not dead :-). My life has just gone crazy.

I believe these are all functions of the synthesizer end of the MIDI
equation.  E.g. MIDI notes are just 0-127.  There's various control change
messages we can send (like CC1, the modulation wheel, or CC4, the "foot
pedal"), but it's up to bs-16i or whatever synth you may have it hooked up
to, to interpret those as something.  I don't think the MIDI device has a
way of directly saying "I want the notes to be played with A432 now".

Of course if you know something I don't here, I'm all ears :-)

SalusaSecondus, 2022-07-21:
That matches what my research has found out in the last day or so. I read the MIDI tuning specification and vaguely understand it. My thread with the bs-16i devs suggests that I just build my scene and things should "just work". So, that's what I'll try next. I just wasn't sure if there was a way for the instrument to signal or control this.

This all started when I some people were asking me to learn a piece with half-flat notes in it. When I mentioned that it wasn't possible their comment was "Don't worry. A normal flat is 'close enough'." But with the ability to actually hack the instrument, I wasn't about to accept that and started digging.

If no changes are needed in the gurdy code, I'll just describe here how I accomplished it and close out the task myself.

bazmonk, 2022-07-21:
bs-16i does have those microtuning settings, yeah.  Something I ran into is
that in wanting to make the transposing aspect as good as possible (I see
that as one of the big strengths of an electronic gurdy, not being confined
to a few keys), anything other than equal temperament gets difficult.  I
couldn't find a way to get the microtuning to easily move around as I
transposed the gurdy up and down.  Each specific transposition on the gurdy
seems to need its own microtuning setup.

SalusaSecondus, 2022-07-24:
I haven't played with transpositions much, but since I can save the microtuning as a scene it's trivial for me to either load a preset which works well enough *or* to create a preset dedicated to it. Thanks to my earlier PR, we can associate tunings with scenes, so this all fits together nicely.

Only 5 minutes ago I played a piece in an Arabic tuning which uses a half-flat (and verified that it worked with a tuner). It worked perfectly.

It looks like there really is no work needed here, so I'm closing this out.

closed 2022-07-24 by SalusaSecondus