bazmonk/digigurdy-baz #82: Getting occasional note stuck on, do we still have the MIDI-all-stop after crank stopped for >3s as in earlier versions?

XenonJohn, 2022-07-31:
I am seeing occasionally that the melody note on Channel 2 gets stuck on even when cranking has stopped and all other channels are silent (using a Teensy 3.5). In older code versions there was a MIDI-kill feature that turned off all 6 channels if cranking had stopped for more than 3 seconds (there is a MIDI cc command that has this function) which stopped this ever becoming really annoying. I wonder has that been taken out of this latest code version 1.4?

Small thing regarding display of channels to avoid any confusion:
Also when you select a preset, it lists them in the order melody, melody, drone, tromp and then once you have actually confirmed your selection, it lists the channels in slightly different order as melody, melody, tromp, drone.  When you set up your iPad running the bs-16i app, you need to set up the first 4 channels as: Ch1 melody,  Ch2 melody, Ch3 tromp, Ch4 drone, i.e. the same way they are listed in the final screen display after you have confirmed the preset you definitely want. It might be best to always list the channel settings in all the OLED screen displays in the order of melody, melody, tromp, drone to avoid any confusion when setting up the bs-16i channels, or even better if space allows Ch1mel, Ch2mel, Ch3Tromp, Ch4Drone or something along those lines.

bazmonk, 2022-09-08:
Getting to this very late...

I don't think my versions of the digigurdy code ever had that.  I can add it... but I'd rather understand why the channel is getting stuck on.  

The big arcade button for crank-less playing does send a kill to all channels, and should function to at least make that noise stop.  If _that_ doesn't stop it, let me know because it really should.

Currently I'm focusing on code for the newer optical cranks, and my digigurdy is in pieces wired up to a breadboard and a hacked-up development version of the optical crank.  I plan to get a full-on optical version as soon as I can, and when I do I will be putting this back into a motor-driven version, and will revisit this bug.

Is there any way you can reproduce or encourage the note to get stuck?  It would help me figure out what's doing it.

I'll also look into making the order of strings more consistent across the board.  Thanks for pointing that out.

XenonJohn, 2022-09-08:
Hi Basil,

It's not a big deal  - as you say if I turn on auto-crank mode with the big
button then off again, it stops it.

Lets see if it is still there after you've done your new (amazing) chopper
wheel all on the Teensy 4.1    code.

BW

John

bazmonk, 2022-11-03:
I switched the strings on that preset screen to the right order just now.  It's not in the 2.0rc1 but it'll be in 2.0rc2 (and it's in the code right now).

I'm closing this issue because if there is a note stuck on issue, it's not due to this.

I think some (rather rare) fake buzzes come from the sensor landing just right on a spoke.  I have a lot of trouble reproducing it and I'm not sure if that's what it is, but for now I'l close this to clean up my issue list :-)

closed 2022-11-03 by bazmonk