bazmonk/digigurdy-baz #85: Crank Detected when no crank attached

Gubbledenut, 2022-10-12:
Not sure how far back this goes, but 1.7.5 unto 1.8, currently have 1.7.8 all say Crank detected on startup with no crank module connected...perhaps this has become obsolete on start up?
Everything else works fine, ie crank connected works, along with crank button to play without..or not connecting crank and using crank button??

bazmonk, 2022-10-12:
Oh yeah, that's known in the optical crank versions.  At first I just forced it to detect to make it easier for me to test, but it doesn't look like John's new ones have a detachable keybox at all, so I just left it that way.

You're saying when the crank isn't there, it still works fine despite claiming the crank is there, right?  Since it's digital I suspect you don't get spurious readings when there's nothing connected.  Haven't checked. 

I'll poke around and look into this.  I think crank detection isn't necessary with a digital crank, *but* since I do want one unified code again eventually that supports both, I think I should leave the routine in there and have it print something else when crank detection isn't needed (rather than gut it out entirely).

bazmonk, 2022-10-12:
I just verified my suspicion is correct: crank detection is just plain not necessary with the optical crank.  You can even hot-swap it!

I'll leave this issue up until I do something about the misleading crank-detection screen itself.  Thanks for bringing this up!

bazmonk, 2022-11-03:
Removed the crank-detection stuff entirely.  Will re-introduce conditionally when the motor crank comes back into play.  Optical cranks can be hot-plugged while playing.

closed 2022-11-03 by bazmonk