bazmonk/digigurdy-baz #88: problem compiling 1.9.3/4

Gubbledenut, 2022-10-24:
Getting compile error do we need to install the "string" library

/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/EEPROM/EEPROM.h:191:8: error: 'String' does not name a type
 inline String &EEPROMClass::get(int idx, String &s){
        ^
Using library EEPROM at version 2.0 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/EEPROM 
Using library MIDI at version 5.0.2 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/MIDI 
Using library ADC at version 9.1 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/ADC 
Using library Bounce in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Bounce (legacy)
Using library Adafruit_GFX_Library at version 1.11.3 in folder: /Users/daj/Documents/Arduino/libraries/Adafruit_GFX_Library 
Using library Adafruit_BusIO at version 1.12.0 in folder: /Users/daj/Documents/Arduino/libraries/Adafruit_BusIO 
Using library Wire at version 1.0 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Wire 
Using library SPI at version 1.0 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/SPI 
Using library Adafruit_SSD1306 at version 2.5.6 in folder: /Users/daj/Documents/Arduino/libraries/Adafruit_SSD1306 
Error compiling for board Teensy 4.1.

bazmonk, 2022-10-24:
Yeah… I messed up.

I was moving menus and ripping stuff out and at some point I had ALL the menus moved off, it compiled well, and that was going to be 1.9.4.  Then I found runtime errors.  I don’t think I can pass around objects like I am.

At any rate, I found the commit where I introduced the problem, and was part-way through backtracking… and I had to sleep.  So the latest commit as-is, not working.

I’ll back out of all those changes, go back to the last fully-working version (it’s in 1.9.3), and try yesterday’s work again.

You won’t need to install new libraries.  When it’s done it’ll take less!

-Baz

bazmonk, 2022-10-24:
I've undone last night's work that introduced this problem.  If you use the latest testing, it should compile now.

v1.9.3's build is the last one I trust.

bazmonk, 2022-10-24:
v1.9.4--without any runtime errors I can find--is out.  The testing branch has it, and it should compile clean now.  I'll leave this open if you want to confirm.

Gubbledenut, 2022-10-24:
tried a couple of times still getting same string in eprom error compiling, also tried the 194 hex and it came up with an error in the reboot window
 "Teensy Loader is unable to read your compiled sketch (r)  This error should never happen.  Please report this to paul@pjrc.com.  In Teensy Loader, click Help > Verbose Info, then click Log > Save As and attach the log file to your message.  The log info is essential for any hope of figuring out what went wrong here!"

bazmonk, 2022-10-24:
Can you paste me the whole compile log when you try to build the latest one?

Gubbledenut, 2022-10-24:
on a Mac  running teensyduino 1.57 direct from teensy  ..which is ever so stable..can try in Arduino IDK too but that runs same teensy 1.57 in the background too??

Arduino: 1.8.19 (Mac OS X), TD: 1.57, Board: "Teensy 4.1, Serial + MIDI, 600 MHz, Fastest, US English"

/Applications/Teensyduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Teensyduino.app/Contents/Java/hardware -tools /Applications/Teensyduino.app/Contents/Java/tools-builder -tools /Applications/Teensyduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Teensyduino.app/Contents/Java/libraries -libraries /Users/daj/Documents/Arduino/libraries -fqbn=teensy:avr:teensy41:usb=serialmidi,speed=600,opt=o3std,keys=en-us -ide-version=10819 -build-path /var/folders/w2/wcq75bks2cn8kctqm2vwvx7c0000gn/T/arduino_build_425803 -warnings=none -build-cache /var/folders/w2/wcq75bks2cn8kctqm2vwvx7c0000gn/T/arduino_cache_892634 -verbose /Users/daj/Music/Instrument Plans/DigiGurdy_electronic_hurdy-gurdy_keybox/DigiCodes_Current/BAZ Versions/digigurdy-baz-194/digigurdy-baz-194.ino
/Applications/Teensyduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Teensyduino.app/Contents/Java/hardware -tools /Applications/Teensyduino.app/Contents/Java/tools-builder -tools /Applications/Teensyduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Teensyduino.app/Contents/Java/libraries -libraries /Users/daj/Documents/Arduino/libraries -fqbn=teensy:avr:teensy41:usb=serialmidi,speed=600,opt=o3std,keys=en-us -ide-version=10819 -build-path /var/folders/w2/wcq75bks2cn8kctqm2vwvx7c0000gn/T/arduino_build_425803 -warnings=none -build-cache /var/folders/w2/wcq75bks2cn8kctqm2vwvx7c0000gn/T/arduino_cache_892634 -verbose /Users/daj/Music/Instrument Plans/DigiGurdy_electronic_hurdy-gurdy_keybox/DigiCodes_Current/BAZ Versions/digigurdy-baz-194/digigurdy-baz-194.ino
Using board 'teensy41' from platform in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr
Using core 'teensy4' from platform in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr
Detecting libraries used...
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++ -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=157 -DARDUINO=10819 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_MIDI_SERIAL -DLAYOUT_US_ENGLISH -I/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/cores/teensy4 /var/folders/w2/wcq75bks2cn8kctqm2vwvx7c0000gn/T/arduino_build_425803/sketch/digigurdy-baz-194.ino.cpp -o /dev/null
Alternatives for Bounce.h: [Bounce]
ResolveLibrary(Bounce.h)
  -> candidates: [Bounce]
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++ -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=157 -DARDUINO=10819 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_MIDI_SERIAL -DLAYOUT_US_ENGLISH -I/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/cores/teensy4 -I/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Bounce /var/folders/w2/wcq75bks2cn8kctqm2vwvx7c0000gn/T/arduino_build_425803/sketch/digigurdy-baz-194.ino.cpp -o /dev/null
Alternatives for MIDI.h: [MIDI@5.0.2]
ResolveLibrary(MIDI.h)
  -> candidates: [MIDI@5.0.2]
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++ -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=157 -DARDUINO=10819 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_MIDI_SERIAL -DLAYOUT_US_ENGLISH -I/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/cores/teensy4 -I/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Bounce -I/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/MIDI/src /var/folders/w2/wcq75bks2cn8kctqm2vwvx7c0000gn/T/arduino_build_425803/sketch/digigurdy-baz-194.ino.cpp -o /dev/null
Alternatives for ADC.h: [ADC@9.1]
ResolveLibrary(ADC.h)
  -> candidates: [ADC@9.1]
/Applications/Teensyduino.app/Contents/Java/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++ -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=157 -DARDUINO=10819 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_MIDI_SERIAL -DLAYOUT_US_ENGLISH -I/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/cores/teensy4 -I/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Bounce -I/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/MIDI/src -I/Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/ADC /var/folders/w2/wcq75b [... cut ...]
Using library Wire at version 1.0 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/Wire 
Using library SPI at version 1.0 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/SPI 
Using library Adafruit_SSD1306 at version 2.5.7 in folder: /Users/daj/Documents/Arduino/libraries/Adafruit_SSD1306 
Using library EEPROM at version 2.0 in folder: /Applications/Teensyduino.app/Contents/Java/hardware/teensy/avr/libraries/EEPROM 
Error compiling for board Teensy 4.1.

bazmonk, 2022-10-24:
I'm in the 1.8.19 Arduino, 1.56 teensyduino.

...And I get your error now with 1.57.  Ok... Let's see...

bazmonk, 2022-10-24:
Here we go!

Teensyduinio 1.57 changelog:

> EEPROM fix get() & put() with String (Luni64)

> EEPROM get() and put() work with String thanks to Luni64.

Thanks Luni64!

So, when you open a `.ino` file in Teensyduino, a lot of stuff "just works" in there, because it loads all this stuff behind the scenes for you.  One of those is `String`, their implementation of a string class.

Well, they added support for that to EEPROM.  Neato!  But that means EEPROM relies on it.  I'm guessing Luni64 doesn't include that in EEPROM.h's header, and no one's noticed because in an `.ino` file, `Arduino.h` is loaded in the background.

So now I have to include String support myself where ever I use EEPROM, which I shouldn't have to.

https://forum.pjrc.com/threads/71359-EEPROM-h-doesn-t-include-String-library-on-its-own-now-that-it-has-String-support?p=314717#post314717

Let's see what they say.

In the meantime, I fixed it.  Try with the latest commit to testing.

Gubbledenut, 2022-10-25:
yup working running 1.9.5 that is so fast booting up! now about getting out of the about screen ??? not even sure how I got there with the option 2 , 1 being nothing as the start up is so fast can't see the ver flash past my eyes....

bazmonk, 2022-10-25:
The X button (top left) backs out of the about screen.  I'll slow down the delay at the intro again...

bazmonk, 2022-10-25:
I'm closing this issue.  Heard back from the Teensyduino folks: this problem is known but since the Arduino environment so heavily assumes the Arduino libraries are loaded, it looks like they're not in a hurry to fix this.  I'll leave in my workaround as basically a requirement now.

closed 2022-10-25 by bazmonk