bazmonk/digigurdy-baz #90: optical crank spoke count

Gubbledenut, 2022-10-24:
Notice you've changed your default no of spokes in config to 80 down from 89?
I'm still running 90 spoke, but can easily print up an 80 spoke if you think that is optimum, or is more better, it's an easy adjust on the config page now???

bazmonk, 2022-10-24:
It shouldn't make a big difference right now.  I switched to 80 because that's what John is using in the "production" ones, and I have one now.  The primary "target" has always been what John's selling, basically.

With the brake design (and the way John's don't reverse crank well at all, to the point it's not even worth thinking about), 80 spokes is fine.  I think the next step would be a all-in-one encoder that gives you position and everything from the get-go without any mucking about.

I'll make it adjustable on the gurdy itself if/when it becomes relevant.

Gubbledenut, 2022-10-25:
cool; I will use 80 too to keep more standard...my friction pully optical set up works fine in both directions...I didn't want to fit more mechanical bits outside the crank module ..ie the drum brake. one day might refine my setup but for now works great, and means it is still modular, so keybox will work off body and unplugged from crank module, foot switch and LEDs..and tsunami one day

bazmonk, 2022-11-03:
Closing this one.  Stickin with 80 in the builds for now.

closed 2022-11-03 by bazmonk