bazmonk/digigurdy-baz #91: knob LED toggle

Gubbledenut, 2022-10-26:
rather than have the led knob hardwired in the code, could it be an option to toggle in the ex buttons or in the other menu??
I have put a latching switch on their ground line to physically turn on or off..but.. that's just me..

bazmonk, 2022-10-27:
Yeah.  I plan to do it like this:

* Whether to enable the LED at all will be a variable in `config.h`, and my builds will have it on by default.
* In the Options menu will be a place to turn it on or off.
* Which one you want on will be saved in EEPROM and loaded on start-up.

This way, it'll be on by default, people can turn it off and it'll remember over restarts, and in situations where you're using that pin and *need* it to be never used as an LED, you can override the default and compile it entirely disabled.

Will that work?

Gubbledenut, 2022-10-27:
yes perfect, I guessed you were going to do something like that.
At the moment I am only using a standard 24 key board , but once we have things sorted will configure the code to run on the other keyboards I've made, the extended 31 key "tenordigigurdy" the harpa and the mora harpa..which are both on hold ..enjoying the 24 key and latest code for now...tsunami will be good...

bazmonk, 2022-10-27:
I'm aiming to make the Tsunami/Trigger take actual MIDI-IN, so it should be
easier to deal with on the hardware side: you'll send it what you already
send to the MIDI jack.  Also thinking of making the MIDI-OUT on the Trigger
side a MIDI-THRU, which can go back to the actual MIDI jack, and... problem
solved.  Just power the Trigger somehow.

Gubbledenut, 2022-10-27:
Power to my tsunami is via a usb c csble, and a usb micro to the teensy from a two outlet 10k mah power brick, both cables have on/off switches…again this is here the super wave trigger difers…
Anyway you can get it to work, just let me have the pin outs…so  that midi will be teensy tx1 to tsunxmi midi rx ….think the line ou pt is where the super wav trigger differs
And yea I got the last one available here too!

bazmonk, 2022-10-28:
If you’re going to power it separately, you’ll just need the Tx1 and a GND.  I *think* that you could even steal from the same VIN that the Teensy itself uses, and get 5V that way to power the Tsunami.  Then it’s… literally the same thing as MIDI-OUT, minus the resistors.

I’m trying to also do it as MIDI (so you’ll use the super Tsunami’s MIDI Rx, not the Serial one that John has been using), because it’s easier on the digigurdy side and won’t require any changes or settings to use one or the other: it’s all MIDI as far as the Teensy is concerned.

Tonight, still powered separately, my goal is to make my spare teensy trigger notes as MIDI, over the Tx1 and GND.  If I can do that, I know I’m off to the races with the rest of it.

I have a feeling this won’t take any work on the Teensy side at all.  To be honest, I’m not sure where John got stuck yet just hooking this up via MIDI.  I haven’t hit any snags except my Mac is too new to use the Tsunami firmware updating app.  But I am using the WAV trigger, the cheaper one that Samuel had a good experience with apparently.

-B

bazmonk, 2022-10-28:
What about the line output?  The WAV Trigger has line-level stereo out via the jack, and if I use the speaker pins it’s got a on-board amp (which I don’t plan on using… line-out to my headphones was surprisingly loud).  Doesn’t the Tsunami have this?

Gubbledenut, 2022-10-28:
Yes single power from the 3.3V from the teensy may work, the super is a 3.3v compliant board,This is from the https://robertsonics.com page. makers of the whole wav trigger and tsunami setup...so jack connected to line output 1 R and 1 L of many on my super board, also wondering if the issue with it over the basic wav trigger is the "improved" midi controller, which is why John ended up going for the serial input instead???

"What's different from the WAV Trigger?
Tsunami is faster and has more memory so that it can play more simultaneous tracks (more voices) and it has either 8 mono or 4 stereo output channels instead of only 1 stereo. Each output can have its own volume and playback rate.

Tsunami can loop tracks seamlessly over any arbitrary file length. It has a dedicated MIDI port with a built-in opt-isolator so no external circuitry is required to connect to a MIDI controller. An extension to the file naming convention allows you to map any MIDI key to any of the 4 audio outputs, making it easy to “split” your keyboard controller.

It has a line-level stereo audio input.

It has a reset button, making it possible to change the SD card with requiring a power cycle. And it has a proprietary boot-loader that provides for firmware updates directly from the microSD card."

bazmonk, 2022-10-28:
You’d be taking the same V-IN as what’s coming off the USB powering the Teensy.  It’s a 3.3V board but it takes 5V+ power.

Here’s the 3.3v-based Teensy4.1, the Trigger piggybacking on the power, all running off a simple 18650 battery (the kind John uses in his gurdies).  Actual sound works to my headphones:

I’ll just need one more cable to go from Tx1-Teensy -> Rx-Trigger, and that’s all she wrote!

bazmonk, 2022-10-28:
Oh, image didn’t work.  
[image]

bazmonk, 2022-10-28:
Might have found the hard part lolSent from my iPhone

Gubbledenut, 2022-10-28:
Awesome that kinda works... I am running my tsunami with a second usb cabe from my power bank, I also think it's a good idea to be able to switch it off if you want to run through either an iPad or use the midi din ???
I piggybacked  pins 4 (VCC)  and 5 (TX) from the back of the 5 pin din to the Miidi input pins 4 and 5 on the tsunami
a 3.5 mm Socket is patched into output 1 L and 1 R and GND
Sound does come out, and it's the correct notes , but not correct sounds or in the right places, ie drones, etc .....over to you...

bazmonk, 2022-11-03:
This is finally coming up.

Stay tuned...

bazmonk, 2022-11-03:
This works for me!

So if you remove `#define LED_KNOB` from `config.h`, the pin isn't touched at all.  We don't make an LED object, we never write to it at all.

If you do have it defined, LED is enabled.  The on/off works without having to restart the gurdy.  It's in the playtime Pause->options menu.  This has the LED object still defined, but we just don't trigger it on/off and stop writing to the pin.

When you change the LED option in the menu, that saves to an EEPROM slot, and that preference is loaded up next time after a restart.

If you clear the EEPROM upon startup (and you should, I've added EEPROM stuff and the clear function sets some defaults for me, like this one), the default is LED on (which does nothing if you have it disabled in `config.h`).

I'll leave the issue open till you try it out but it seems solid.  If you wanna test, double-check for me that if LED is disabled, attempting to enable/disable it via the menu simply does nothing (it's as if you didn't press the button).

Gubbledenut, 2022-11-03:
That  ver 2.0cr1 works very nicely indeed, but I did  clear the eeprom first time install the ex buttons didn't work...were fine in previous.
I did have a slot saved, and I did uncomment one of the string screen tsunami midi text??
while it might be a pain, worth suggesting in instructions if there is any weird behaviour clear the eeprom....???

bazmonk, 2022-11-03:
I did mention that on facebook and in the README after replying here,
yeah.  The values before clearing (when they were cleared before) would be
0, and the ex buttons don't have a 0 function.

The clear_eeprom() now sets the the EX1/2/3 functions to 1,2,and 3.  It
also enables the buzz (without clearing, the buzz LED wouldn't work).

Thanks for lookin' out!  Can I close this issue as solved?

Gubbledenut, 2022-11-03:
yes close it ..thanks

closed 2022-11-03 by bazmonk